LibreELEC/LibreELEC.tv #8595: [BUG] [librespot] [v11.80.5.0.]

beralt85, 2024-02-06:
### Describe the bug
When using the librespot service, the kodi machine shows up as a device in the spotify apps on mobile or desktop, but connecting to the device does not work.

### How to reproduce
Steps to reproduce the behavior:
1. Install librespot from official librelec addon repo
2. From a spotify app, try to connect to the librelec device.
3. See error

### Information
 - LibreELEC Version: [nightly-20240204-78a0ba1] (Kodi 21.0-BETA2) (20.90.822) (but similar on other versions)
 - Hardware Platform: [e.g. RPi5]

### Fix
There is a fix for that described by vpeter in the coreelec forum: https://discourse.coreelec.org/t/s905x2-librespot-doesnt-work/7556/9
This works also on libreelec:

> This command should be executed before running librespot
> 
> `echo 2 >/proc/sys/abi/cp15_barrier`
> 
> So the solution is adding this line to the file /storage/.config/autostart.sh. If this file doesn’t exist create it. And the file muct be in Unix format (don’t use Windows Notepad to create this file).
> 
> New versions of this addon will have this workaround included.

heitbaum, 2024-02-07:
@henri-funck / @vpeter4 - could you please take a look. Not sure why the 2020 "workaround" is required again?

chewitt, 2024-02-07:
Librespot is working fine for me here with a current (within the last couple of days) LE12 image and an RPi5.

beralt85, 2024-02-27:
I just tested on a new install.
`/proc/sys/abi/cp15_barrier` was already set to `2` and it connects.
At the moment, I have the problem that while it connects and plays (meaning that the song proceeds, and I can pause/resume it) I don't get any audio.
But this seems to be a different issue. So I would close this one now.
Thanks for the replies!

closed 2024-02-27 by beralt85