LibreELEC/LibreELEC.tv #8609: [BUG]   evergreen_surface_check_linear_aligned:214 texture pitch 1920 invalid must be aligned with 256

oliv3r, 2024-02-10:
When playing back hevc 10bit video, I get the following in my kernel log. There is only audio and the video just garbles whatever was in the framebuffer a bit.

```
[   51.686234] radeon 0000:01:00.0: evergreen_surface_check_linear_aligned:214 texture pitch 1920 invalid must be aligned with 256
[   51.686246] radeon 0000:01:00.0: evergreen_cs_track_validate_texture:829 texture invalid 0x1dfc3bc1 0x10000323 0x0b200000 0x00000000 0x00000000 0x80020001
[   51.686252] [drm:radeon_cs_ioctl] *ERROR* Invalid command stream !
```

This is on the nightly of last week; I'll try todays nightly tomorrow, but have strong feeling it will remain

chewitt, 2024-02-11:
Put Kodi in debug mode and reboot then demo the problem and run "pastekodi" and share the URL generated. Playback content isn't so useful but Kodi also includes a load of environment and hardware detail that will be relevant.

The same error shows up in other bug reports which give some hints in these about the issue being with mesa and maybe influenced by the smorgåsbord of old/new mesa drivers/options that could be used by Kodi. Most of them seem to relate to systems with with dual GPUs:

https://github.com/ValveSoftware/steam-for-linux/issues/8508
https://gitlab.freedesktop.org/mesa/mesa/-/issues/6227
https://gitlab.freedesktop.org/mesa/mesa/-/issues/6326

oliv3r, 2024-02-11:
Is there an easy way to strip out filenames :) Last time I spent 10 minutes manually removing filenames :p

oliv3r, 2024-02-13:
So, Here's the non-debug log with yesterday's nightly. The dmesg errors have gone, so that's good. Also the weird image refresh of the buffers. The video output is still blank, only audio.

```
2024-02-13 18:28:40.853 T:1144     info <general>: VideoPlayer::OpenFile: nfs://11080p BluRay x265 10bit.mkv
2024-02-13 18:28:40.859 T:1269     info <general>: Creating InputStream
2024-02-13 18:28:40.962 T:1269     info <general>: Creating Demuxer
2024-02-13 18:28:40.994 T:1269     info <general>: Opening stream: 0 source: 256
2024-02-13 18:28:40.994 T:1269     info <general>: [WHITELIST] Searching the whitelist for: width: 1920, height: 804, fps: 23.
976, 3D: false
2024-02-13 18:28:40.994 T:1269     info <general>: Display resolution ADJUST : 1920x1080 @ 23.976025 Hz (28) (weight: 0.000)
2024-02-13 18:28:41.007 T:1144     info <general>: VideoPlayer: OnLostDisplay received
2024-02-13 18:28:41.007 T:1144  warning <general>: CDVDMessageQueue(audio)::Put MSGQ_NOT_INITIALIZED
2024-02-13 18:28:41.007 T:1144  warning <general>: CDVDMessageQueue(video)::Put MSGQ_NOT_INITIALIZED
2024-02-13 18:28:41.007 T:1268  warning <general>: CVideoSyncGbm::Run: drmCrtcGetSequence failed (-1)
2024-02-13 18:28:41.007 T:1268  warning <general>: CVideoSyncGbm::Setup: drmCrtcGetSequence failed (-1)
2024-02-13 18:28:41.091 T:1144     info <general>: GLES: Maximum texture width: 16384
2024-02-13 18:28:41.092 T:1269     info <general>: Creating video codec with codec id: 173
2024-02-13 18:28:41.092 T:1269     info <general>: CDVDVideoCodecDRMPRIME::Open - using decoder HEVC (High Efficiency Video Co
ding)
2024-02-13 18:28:41.104 T:1269     info <general>: Creating video thread
2024-02-13 18:28:41.104 T:1276     info <general>: running thread: video_thread
2024-02-13 18:28:41.104 T:1269     info <general>: Opening stream: 1 source: 256
2024-02-13 18:28:41.104 T:1269     info <general>: Finding audio codec for: 86018
2024-02-13 18:28:41.108 T:1269     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
2024-02-13 18:28:41.108 T:1269     info <general>: Creating audio thread
2024-02-13 18:28:41.108 T:1277     info <general>: running thread: CVideoPlayerAudio::Process()
2024-02-13 18:28:41.108 T:1269     info <general>: Opening stream: 2 source: 256
2024-02-13 18:28:41.131 T:1144     info <general>: VideoPlayer: OnResetDisplay received
2024-02-13 18:28:41.160 T:1272    error <general>: CDVDVideoCodecDRMPRIME::GetFormat - no supported pixel formats: vaapi, yuv4
20p10le
2024-02-13 18:28:41.162 T:1277     info <general>: CVideoPlayerAudio: display reset occurred, checking for passthrough
2024-02-13 18:28:41.164 T:1277     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
2024-02-13 18:28:41.164 T:1277     info <general>: Creating audio stream (codec id: 86018, channels: 6, sample rate: 48000, no
 pass-through)
2024-02-13 18:28:41.178 T:1167     info <general>: CActiveAESink::OpenSink - initialize sink
2024-02-13 18:28:41.192 T:1276    error <general>: CDVDVideoCodecDRMPRIME::AddData - send packet failed: Operation not permitt
ed (-1)
2024-02-13 18:28:41.232 T:1276    error <general>: CDVDVideoCodecDRMPRIME::GetPicture - videoBuffer:nullptr format:yuv420p10le

2024-02-13 18:28:41.232 T:1276    error <general>: CDVDVideoCodecDRMPRIME::AddData - send packet failed: Invalid data found wh
en processing input (-1094995529)
2024-02-13 18:28:41.250 T:1276    error <general>: CDVDVideoCodecDRMPRIME::GetPicture - videoBuffer:nullptr format:yuv420p10le

2024-02-13 18:28:41.370 T:1167     info <general>: Skipped 26 duplicate messages..
2024-02-13 18:28:41.370 T:1167     info <general>: CAESinkALSA::Initialize - Attempting to open device "@"
2024-02-13 18:28:41.373 T:1276    error <general>: CDVDVideoCodecDRMPRIME::GetPicture - videoBuffer:nullptr format:yuv420p10le

2024-02-13 18:28:41.377 T:1167     info <general>: CAESinkALSA::Initialize - Opened device "surround51"
2024-02-13 18:28:41.377 T:1167     info <general>: CAESinkALSA::InitializeHW - Unable to open the required number of channels
2024-02-13 18:28:41.377 T:1167     info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, tr
ying other formats
2024-02-13 18:28:41.377 T:1167     info <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S32NE
2024-02-13 18:28:41.378 T:1276    error <general>: CDVDVideoCodecDRMPRIME::GetPicture - videoBuffer:nullptr format:yuv420p10le

2024-02-13 18:28:41.850 T:1183     info <general>: Skipped 87 duplicate messages..
```

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share a debug enabled pastekodi. Alternatively please raise this on the forum for support.

closed 2024-03-16 by heitbaum