LibreELEC/LibreELEC.tv #8617: [BUG] No audio over hdmi nvidia ion based hardware

cracklejoy, 2024-02-14:
Trying to play any audio via hdmi results in no sound output.  

All these tests are on legacy hardware images with appropriate sound card selections within kodi.

LibreElec v10 and under has no issues.  

Audio output can be achieved via headphone and spdif out in recent versions but again, not over hdmi.  This has been ongoing for some time with no resolution.  

Some simple tests like  "speaker-test --channels 2 --rate 48000 --device hw:0,3" outputs no sound on versions > 10 while the same test works on versions < 10.

I'm starting to wonder if this is a kernel issue or some larger nvidia and more specificially ion issues.  Regardless, this box does not output audio over hdmi like it used to under libreelec versions greater than version 10. Nothing has changed in my setup except for an updated version of libreelec ie. versions > 10 which obviously does not work as it used to.

All tests were conducted on fresh installs with the same no audio over hdmi issue.

I'm not sure where to go from here as everything looks as though it should be working yet it does not.

[debug log v11.0.6](https://paste.libreelec.tv/valid-rooster.log)
[debug log v12.0-nightly-20240208-f3809d8](https://paste.libreelec.tv/notable-anemone.log)

heitbaum, 2024-03-11:
There has been an update to nvidia since this was raised. Note I’m not sure about  Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share a debug enabled pastekodi and raise this on the forum for support - more people to review it. There are some ION discussions.

closed 2024-03-11 by heitbaum

cracklejoy, 2024-03-11:
For the record, the changes in your mentioned version do nothing as I expected that outcome before I tested this for the last time.  This issue has been ongoing for over a year now without any changes in sight and I don't expect anything to change.  No movement over on the forum either if that even matters as those threads are even more stale.  Nviida Ion support has been dead for awhile now so at least people who may read this in the future can move on.  The "won't fix" closed status seals the deal.  Good luck to the project and it's time to try new hardware for those with ion based machines.  Cheers.

nopb01, 2024-03-12:
Too sad to hear this.
Ion is a great platform for full-hd and I see no reason to replace well working hardware. I will have to stay on LE10 if this bug isn't corrected, and I'm probably not alone.

chewitt, 2024-03-16:
For reference: https://forum.libreelec.tv/thread/27015-libreelec-generic-legacy-v11-0-1-x86-no-audio/?postID=189886#post189886

Discussion continues in the forum, not here.