LibreELEC/LibreELEC.tv #8620: Usb errors since 12.0-nightly-20240115-923fe4e

Nachtzuster, 2024-02-14:
### Describe the bug
I have a ASUS Xonar U7 connected to my RPi4 that keeps resetting since `LibreELEC-RPi4.aarch64-12.0-nightly-20240115-923fe4e`. In `LibreELEC-RPi4.aarch64-12.0-nightly-20240112-fa5abde` it is still fine.
On current nightly (`LibreELEC-RPi4.aarch64-12.0-nightly-20240213-d888148`) it still resetting 
The resets only happen when I play a movie, then it reset every 30 second or so

I rebuild 923fe4e785 and this build gives usb erros too
Then reverting https://github.com/LibreELEC/LibreELEC.tv/pull/8513: This build is fine.

dmesg snippet when it resets:
```
[   17.169701] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   22.679231] Bluetooth: RFCOMM TTY layer initialized
[   22.679265] Bluetooth: RFCOMM socket layer initialized
[   22.679286] Bluetooth: RFCOMM ver 1.11
[   52.330686] usb 1-1: clear tt 1 (0020) error -71
[   52.337267] usb 1-1-port4: cannot reset (err = -71)
[   52.337588] usb 1-1-port4: cannot reset (err = -71)
[   52.337912] usb 1-1-port4: cannot reset (err = -71)
[   52.338211] usb 1-1-port4: cannot reset (err = -71)
[   52.338531] usb 1-1-port4: cannot reset (err = -71)
[   52.338546] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[   52.338842] usb 1-1-port4: cannot disable (err = -71)
[   52.339172] usb 1-1-port4: cannot reset (err = -71)
[   52.339462] usb 1-1-port4: cannot reset (err = -71)
[   52.339794] usb 1-1-port4: cannot reset (err = -71)
[   52.340092] usb 1-1-port4: cannot reset (err = -71)
[   52.340416] usb 1-1-port4: cannot reset (err = -71)
[   52.340433] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[   52.340711] usb 1-1-port4: cannot disable (err = -71)
[   52.341042] usb 1-1-port4: cannot reset (err = -71)
[   52.341347] usb 1-1-port4: cannot reset (err = -71)
[   52.341692] usb 1-1-port4: cannot reset (err = -71)
[   52.343652] usb 1-1-port4: cannot reset (err = -71)
[   52.343985] usb 1-1-port4: cannot reset (err = -71)
[   52.343995] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[   52.344290] usb 1-1-port4: cannot disable (err = -71)
[   52.344572] usb 1-1-port4: cannot reset (err = -71)
[   52.344825] usb 1-1-port4: cannot reset (err = -71)
[   52.345077] usb 1-1-port4: cannot reset (err = -71)
[   52.345331] usb 1-1-port4: cannot reset (err = -71)
[   52.345578] usb 1-1-port4: cannot reset (err = -71)
[   52.345586] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[   52.345872] usb 1-1-port4: cannot disable (err = -71)
[   52.346166] usb 1-1-port4: cannot disable (err = -71)
[   52.346458] hub 1-1:1.0: hub_ext_port_status failed (err = -71)
[   58.759986] usb 1-1.4: USB disconnect, device number 3
[   58.970380] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   59.411490] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
[   59.916432] usb 1-1.4: not running at top speed; connect to a high speed hub
[   59.937153] usb 1-1.4: New USB device found, idVendor=1043, idProduct=857c, bcdDevice= 1.10
[   59.937176] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   59.937181] usb 1-1.4: Product: Xonar U7
[   59.937185] usb 1-1.4: Manufacturer: ASUS
[   60.060837] input: ASUS Xonar U7 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.4/0003:1043:857C.0002/input/input1
[   60.118404] hid-generic 0003:1043:857C.0002: input,hidraw0: USB HID v1.00 Device [ASUS Xonar U7] on usb-0000:01:00.0-1.4/input4
LibreELEC:~ #
```

```
2024-02-14 16:07:58.646 T:1169     info <general>: VideoPlayer::OpenFile: videodb://movies/titles/1172?xsp=%7b%22limit%22%3a15%2c%22order%22%3a%7b%22direction%22%3a%22descending%22%2c%22ignorefolders%22%3a0%2c%22method%22%3a%22lastplayed%22%7d%2c%22rules%22%3a%7b%22and%22%3a%5b%7b%22field%22%3a%22inprogress%22%2c%22operator%22%3a%22true%22%2c%22value%22%3a%5b%5d%7d%5d%7d%2c%22type%22%3a%22movies%22%7d
2024-02-14 16:07:58.647 T:1953     info <general>: Creating InputStream
2024-02-14 16:07:59.332 T:1953     info <general>: Creating Demuxer
2024-02-14 16:07:59.343 T:1953     info <general>: Opening stream: 0 source: 256
2024-02-14 16:07:59.343 T:1953     info <general>: Creating video codec with codec id: 27
2024-02-14 16:07:59.344 T:1953     info <general>: CDVDVideoCodecDRMPRIME::Open - using decoder V4L2 mem2mem H.264 decoder wrapper
2024-02-14 16:07:59.358 T:1953     info <general>: Creating video thread
2024-02-14 16:07:59.359 T:1955     info <general>: running thread: video_thread
2024-02-14 16:07:59.359 T:1953     info <general>: Opening stream: 1 source: 256
2024-02-14 16:07:59.359 T:1953     info <general>: Finding audio codec for: 86018
2024-02-14 16:07:59.360 T:1953     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
[... 534 lines cut ...]
2024-02-14 16:10:43.910 T:1176  warning <general>: ActiveAE - large audio sync error: -1721.059739
2024-02-14 16:10:43.974 T:1176  warning <general>: ActiveAE - large audio sync error: -1736.136184
2024-02-14 16:10:44.039 T:1176  warning <general>: ActiveAE - large audio sync error: -1749.481258
2024-02-14 16:10:44.071 T:1176  warning <general>: ActiveAE - large audio sync error: -1732.866202

```

### How to reproduce
Steps to reproduce the behavior:
1. Play movie
2. audio drops out
3. audio returns
4. audio drops out

### Information
 - LibreELEC Version: LibreELEC-RPi4.aarch64-12.0-nightly-20240115-923fe4e
 - Hardware Platform: RPi4,  ASUS Xonar U7

### Log file
https://paste.libreelec.tv/flexible-lionfish.log

### Context
Interestingly, the resets do not occur when playing normal music (mp3) or stream a YouTube clip.
The movies I play, where I see the resets, are DVD rips, so audio is 48kHz
I resampled an mp3 to 48kHz, and playing that one, I do see the resets - so something 48kHz?

heitbaum, 2024-02-14:
A quick google.
- https://forums.balena.io/t/pulse-audio-in-belena-sound-stop-audio-usb-dac/360053
- https://forums.balena.io/t/balena-sound-dac-disconnects-and-doesnt-reconnect/350378
- https://raspberrypi.stackexchange.com/questions/112672/pulseaudio-alsa-sink-usb-audio-alsa-util-c-got-pollnval-from-alsa

would be worth following the checks these guys did to isolate it as same/similar/different..
- enough power?
- usb3
- cable
- before looking into debug of pulse audio

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images? (Not expecting any change - but good to confirm) There is a report that a U5 with RPi5 is working. Alternatively please raise this on the forum for support.

Nachtzuster, 2024-03-11:
update: I installed Generic-legacy.x86_64-12.0-nightly-20240311-86dba4b
the rate gets flipped to 48k, as expected, but unlike in the RPi4, the U7 is ok.
```
LibreELEC:~ # cat /proc/asound/card2/pcm0p/sub0/hw_params
access: MMAP_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 48000 (48000/1)
period_size: 44100
buffer_size: 88200
```

heitbaum, 2024-03-16:
@CvH could you check yours?

heitbaum, 2024-11-04:
Is this still occurring?

Nachtzuster, 2024-11-09:
Yes, I tried https://test.libreelec.tv/13.0/RPi/RPi4/LibreELEC-RPi4.aarch64-13.0-nightly-20241103-f662bea.img.gz, it does still occur.