LibreELEC/LibreELEC.tv #8622: [BUG]  11.0.3 -> 11.0.6 partially broke NFS (so it actually takes down the LAN..)

bjornstromberg, 2024-02-15:
### Describe the bug
play files that are located on NFS shares (playlist with mp3's was the first way it broke, then when moving to a remote folder to play a video, well that works sometimes)

the loading goes bananas and the network goes down, if i poweroff the htpc (nuc gen 6) the network is still swamped, if i restart the router (power off, power on..), network comes back up.

and will be fine until next try to use a nfs shared item that is not played frequently (it would seem that hot cached files on server side limit the fallout of this issue)

if i ssh the files over to the disk on the nuc they play just fine, and it worked just fine until i manually allowed the upgrade to 11.0.6, i held it on 11.0.3 due to people complaining about NFS problems on 11.0.4, then forgot about that..

current question, how do i get the old download image to roll back to 11.0.3 legacy? i need that link yesterday..

if that is not possible i'll reinstall and skip libreelec. NFS is a core dependency it should not be allowed to break, and especially not 3 patch releases later, still be active.

### How to reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Information
 - LibreELEC Version: 11.0.6 (legacy)
 - Hardware Platform: Nuc Gen 6

### Log file
none

### Context
none

bjornstromberg, 2024-02-15:
alpine linux + kodi it is then.

p1nkl0bst3r, 2024-03-04:
In case anyone else is looking for it, you can find the old image [here](https://archive.libreelec.tv/archive/LibreELEC-Generic-legacy.x86_64-11.0.3.img.gz)