LibreELEC/LibreELEC.tv #8626: Continuation of battery debate. Let’s see what your actual users have to say

GavinDarkglider, 2024-02-15:
Continuation of: https://github.com/LibreELEC/LibreELEC.tv/pull/8603

Since you don’t think the debate is going anywhere, and if the patch was the issue, I would agree, but the issue is not including upower, and not allowing me to fix it before rejecting it, and calling <512K bloat. I plan on pushing the kodi patches to kodi when it is done. Lakka doesn’t use upower, so pushing that there is dumb. Also what is the point in attempting to contribute if you will never see the point. So at this point, I will put this here for further discussion from users. I am sorry I put an unfinished but working patch file in my pr to fix a minor issue in existing code, while I work on expanding it further to push it upstream. Also, if maintenance was actually the issue, I have already said I would maintain my changes long term.

theofficialgman, 2024-02-15:
> 10% x86

Of which anywhere from 55% to 65% are laptops (which have a battery). So that constitutes a minimum of 5.5% of your userbase.

https://www.cloudavize.com/laptop-vs-desktop/#:~:text=When%20comparing%20the%20number%20of,units%20(61%25%20more%20laptops)
_2010: 201 million laptop units, 157 million desktop units (24% more laptops)
2019: 166 million laptop units, 88.4 million desktop units (61% more laptops)_

> about 80% RPi ... and 10% other ARM SBCs/boxes.
> As neither RPi nor the ARM SBCs are portable devices with a battery you can easily imagine that adding that feature would add unnecessary bloat to 90% of our userbase.

This statement is simply baseless. There are many handheld raspberry pi and other ARM SBC projects for sale and custom built. Many of which are features on raspberrypi.com https://www.raspberrypi.com/news/piboy-mini-just-add-a-raspberry-pi-and-youve-got-a-handheld-retro-gaming-system/ . It is hard to put a number to these users but given that 90% of your userbase is made up by them, even if only 5% had a battery connected (a conservative estimate), that still means an additional 4.5% are affected.

Bringing to total to a VERY conservative 10% of your userbase. HUGE.

[image]

I can't imagine how a feature standard on literally every other operating system that exists (being able to read the battery level on devices with a batter) would be considered bloat.

Consider that Windows XP/Vista/7/8/8.1/10/11, MacOS (since the first portable one), iOS, Android, ChromeOS, All Linux Desktops (I won't bother listing them out) ALL have battery monitoring as a standard feature. LibreELEC is the outlier, not the norm.

Also consider that your software (LibreELEC) is for Audio/Video playback as you say from mostly Home Theater PCs. Don't you think that most of these users are using wireless control devices and audio devices (eg: bluetooth headphones, game controllers, keyboards, mice, and remotes). Many of these devices have no way of reading the battery level from the device side, only through software on the host. Nearly all of them support showing battery info via upower and without it on LibreELEC there is no way to know that your battery is running low or its current status is until it does (bad user experience).

HiassofT, 2024-02-15:
Please use the forum for feature requests.

We totally agree that having the a feature in kodi to monitor battery status of wireless headphones, remotes, controllers etc and getting alerts on critical battery level would be very handy and once that's implemented in kodi we'll revisit that topic.

But until that's done we won"t add upower as x86 laptops aren't something we actively support

closed 2024-02-15 by HiassofT