LibreELEC/LibreELEC.tv #8669: [BUG] After manually switching network interfaces, UI hangs for several minutes

rkershenbaum, 2024-02-28:
### Describe the bug
After manually switching network interfaces, the LE user interface hangs for about three minutes

### How to reproduce
Steps to reproduce the behavior:
1. Go to settings / LibreELEC / network / connections
2. Disable the active network interface, e.g. eth0
3. Enable a secondary network interface, e.g. wlan0
4. After about ten seconds, note that the UI is unresponsive
5. After about three minutes, note that the UI again becomes responsive

### Information
 - LibreELEC Version: LE 12 nightly-20240228-b9f1ab1
 - Hardware Platform: Rpi4, Rpi5

### Log file
[https://paste.libreelec.tv/humorous-hornet.log]

### Context
The o/s is not hung, as ssh operates normally during the LE UI hang.

heitbaum, 2024-03-11:
Ref: https://forum.libreelec.tv/thread/28119-usb3-to-gigabit-ethernet-adapter-on-rpi4-5-troubles/

antonlacon, 2024-03-30:
I can't replicate on a RPi5 and I see banned addons + repo in your log. Reopen after purging the piracy addons with a new debug log.

closed 2024-03-30 by antonlacon