LibreELEC/LibreELEC.tv #8709: [BUG] Nuc comes (sometimes) up with 100 % load (kodi)

eldersixpack13, 2024-03-07:
### Describe the bug
since some days I sometimes see on my Nuc13  that kodi is running with > 100 % load (htop). 
happens in ~1 of ~5 boots
Nuc was just booted, nothing more; no TV or Movie run

### How to reproduce
Steps to reproduce the behavior:
1. check via command line with htop

### Information
 - LibreELEC Version: LE12 with last nightly
 - Hardware Platform: Nuc13
 - fresh LE installation ~10 days ago

### Log file
https://paste.libreelec.tv/primary-caiman.log

screenshot:
[image]

### Context
 @heitbaum, as an also owner of an Nuc, could verify this or is it just my box

heitbaum, 2024-03-07:
Nothing untoward here. I’m running Omega RC1 but otherwise I wouldn’t compare my image (lots of other stuff as wip)

eldersixpack13, 2024-03-07:
it seems I got it: 
Nuc comes with > 100 % load if the monitor is powered off (headless boot ?)

headless triggers:
CWinSystemGbm::InitWindowSystem - failed to initialize Atomic DRM _and_
CWinSystemGbm::InitWindowSystem - failed to initialize Legacy DRM _and_
failed to duplicate EGL fence fd (EGL_SUCCESS)

tried with/without monitor powered on ~5 times

I close this bug

eldersixpack13, 2024-03-07:
error found

closed 2024-03-07 by eldersixpack13