LibreELEC/LibreELEC.tv #8727: [BUG] don't reject "weak" passwords during setup

dilinger, 2024-03-15:
### Describe the bug
During setup of a new libreelec installation, I enabled remote ssh access. I was informed of the default password, and the setup recommended that I change it from the default. Fine, that's a great idea! Unfortunately the password that I wanted to use was too "weak", and was rejected. The other two passwords that I tried were also considered too weak, and were rejected. So I left the password as the default ("libreelec").

### How to reproduce
Steps to reproduce the behavior:
1. Install a libreelec disk image (zcat https://releases.libreelec.tv/LibreELEC-RPi4.arm-11.0.6.img.gz > /dev/mmcblk0) onto a microsd card, and stick the card into a pi4.
2. Boot up the pi, and go through the setup
3. When asked if you want ssh and samba remote access enabled, disable samba but enable ssh
4. When prompted to change the password, say yes and type in a password that's 7-10 characters in length and includes only lowercase letters.
5. Watch as it tells you that the password is too weak, and refuses to set it.

### Information
 - LibreELEC Version: 11.0.6
 - Hardware Platform: RPi4

### Log file

### Context
What's weaker than a weak password? A default password. In trying to encourage a more secure libreelec installation, this actually makes security *worse*, as it incentivizes users to just use the default password.

In setting a root password for libreelec, I don't have the convenience of a browser password manager, so I'm not going to use an autogenerated password. My plan was to set a commonly used home appliance password (since I'm behind a NAT, it's not a big deal), and then later on disable ssh password logins and install an ssh key.

chewitt, 2024-03-16:
Closing because the system is working as designed, thus there is no bug (and this is a bug tracker). If you want to make a feature request please use the forum.

closed 2024-03-16 by chewitt

dilinger, 2024-03-16:
lol, this is a UX bug. It is not a feature request.