LibreELEC/LibreELEC.tv #8736: [BUG] LE12 Beta1: missing custom firewall/iptables settings

fredrik-eriksson, 2024-03-19:
### Describe the bug
Firewall setting in LE12 Beta1 lists three options: "Off", "Home" and "Public". Notably the "Custom" firewall setting (still referenced in the help text for this setting) is missing. 

### How to reproduce
Steps to reproduce the behavior:
1. Go to Settings -> LibreELEC -> Network -> Firewall
2. Try to find the "Custom" firewall setting

### Information
 - LibreELEC Version: 12 Beta1
 - Hardware Platform: RPi5

### Log file
None

### Context
This is a fresh installation.

heitbaum, 2024-03-19:
Hi @fredrik-eriksson 

I believe you are looking for this:
`NF_CUSTOM_PATH = "/storage/.config/iptables/"`
https://github.com/LibreELEC/service.libreelec.settings/blob/3d3773271cd762f159caf1d12cea65b79763c07a/resources/lib/modules/connman.py#L592-L610

`custom` will be available when the custome rules.v4 is in the directory.

heitbaum, 2024-03-19:
Closing as not bug. Forum is the best for support.

closed 2024-03-19 by heitbaum

fredrik-eriksson, 2024-03-20:
Sorry for the noise, turned out to be a corrupted file system that had emptied everything under .config. Thanks for the pointer!