LibreELEC/LibreELEC.tv #8760: Wetek Play 2: Invalid-key while connecting to wifi

rdorsch, 2024-03-24:
### Describe the bug
This is a continuation of issue #7166 but for LE12.

### How to reproduce
Steps to reproduce the behavior:
1. Install LibreELEC-AMLGX.aarch64-12.0-nightly-20240323-353213e-box.img.gz on Wetek Play 2
2. Try to connect to a Wifi network
3. Consistently an invalid-key message is obtained

### Information
 - LibreELEC Version: 20240323-353213e
 - Hardware Platform: Wetek Play 2

### Log file
kodi.log:
https://paste.libreelec.tv/in-elephant.log

dmesg output:
https://paste.libreelec.tv/welcomed-sloth.log

### Context
I have LE12 on rock64 working flawless on the same OpenWrt router. The wireless signal strength reported by the wetek is rather low (~50), but also with the wifi router next to the wetek (~78) the same error occurs.

What is weird: with an Android Wifi-Hotspot with similar signal strength (~76), wifi on the wetek play 2 works flawless.

heitbaum, 2024-03-25:
I can see the following in the logs. The WPA3 error also in the 8731 report.

iwd[371]: Could not register frame watch type 00b0: -22
and WPA3 (CMD_SET_CQM failed line) the line before “connect failed”

jonian, 2024-03-25:
I'm having the same issue on RPi5 with LibreELEC 12 Beta 1. Had the same issue with LE 11 and updated to 12 in hope that it will be fixed. Irrelevant to this issue, but now I'm also having issues with BT connectivity and a DS4 controller.

vilhelmgray, 2024-03-30:
I'm curious if you are able to connect to your OpenWrt router when configured for an unencrypted connection. In issue #8731, I seem to have trouble with unencrypted signals too which seems to indicate that the problem lies elsewhere in `iwd` rather than the security configuration. I suspect, as heitbaum pointed out, that the `invalid-key` error message is just a default failure code for any problem the arises during that point.

heitbaum, 2024-11-04:
Is this still occurring with the latest LE13 nightlies?
Are the following items (known issues) addressed:
- Is FT off
- Is the SSID not WPA3
- does the wireless device have good signal strength

rdorsch, 2024-11-11:
I hope, I manage to give it a try next weekend.

rdorsch, 2024-11-16:
Am Montag, 4. November 2024, 17:42:50 MEZ schrieb Rudi Heitbaum:
> Is this still occurring with the latest LE13 nightlies?
> Are the following items (known issues) addressed:
> - Is FT off
yes
> - Is the SSID not WPA3
SSID is DLAN
Wireless security is WPA2-PSK (CCMP)

Router is D-Link DIR-505 running OpenWrt 23.05.5 with radio Atheros AR9330 
802.11b/g/n

> - does the wireless device have good signal strength
yes

Unfortunately I still see the invalid-key error:

dmesg output:
https://paste.libreelec.tv/honest-ibex.log

kodi.log:
https://paste.libreelec.tv/composed-turtle.log

Output of journalctl
https://bokomoko.de/~rd/Libreelec/jounalctl.log

I hope that help.

Rainer

heitbaum, 2024-11-16:
A recent lore article - similar to yours - status 16.

https://lore.kernel.org/all/e19e7527-9cc1-4c84-95ac-dcb39f836d58@gmail.com/

would need some debug logs and review of that thread.