LibreELEC/LibreELEC.tv #8762: I'm having the same issue on RPi5 with LibreELEC 12 Beta 1. Had the same issue with LE 11 and updated to 12 in hope that it will be fixed. Irrelevant to this, but now I'm also having issues with BT connectivity and a DS4 controller.

heitbaum, 2024-03-26:
I'm having the same issue on RPi5 with LibreELEC 12 Beta 1. Had the same issue with LE 11 and updated to 12 in hope that it will be fixed. Irrelevant to this, but now I'm also having issues with BT connectivity and a DS4 controller.

_Originally posted by @jonian in https://github.com/LibreELEC/LibreELEC.tv/issues/8760#issuecomment-2018014154_

heitbaum, 2024-03-26:
@jonian - I have created this new issue for you. The issue reported in 8760 will not be the same. For us to help with this "bug" - you will need to run a full debug and investigation. See https://github.com/LibreELEC/LibreELEC.tv/issues/7166#issuecomment-2001716879 as a start. For the BT and DS4 issue - please raise on the forum for support. These are not WiFi related.

geerlingguy, 2024-05-22:
I'm currently having this issue with 12.0.0 on my Pi 5 and a 5GHz network. 2.4GHz network seems fine for now, after I had the 5G network issues. The first boot, I connected to my 5GHz WiFi network, and it worked fine. It seemed like it may have been downloading some updates in the background while I was testing the UI out a bit.

After a later reboot, I could no longer access the Pi over my network so I looked in the LibreELEC settings and sure enough, the 'State' of the Connection was 'failed'. I tried reconnecting but kept getting the 'invalid-key' error.

heitbaum, 2024-11-04:
Is this still occurring with the latest LE13 nightlies?
Are the following items (known issues) addressed:
- Is FT off
- Is the SSID not WPA3
- does the wireless device have good signal strength

damnms, 2025-01-04:
Still occurs, yes. What i also noticed, in the overview where i see the connections, sometimes the IP is not listed even tho the device is connected

jonian, 2025-01-05:
Forgot to report back on this. I had the wifi issue (among other issues) when I upgraded from LE11 to 12. The issue is not present for me on a clean install of LE12.

heitbaum, 2025-01-05:
Thanks for confirming. Closing.

closed 2025-01-05 by heitbaum