LibreELEC/LibreELEC.tv #8818: [BUG]

tgiesela, 2024-04-18:
### Describe the bug
After installing Inputstream Adaptive (v21.4.5.1) video's using the widevine DRM do not play due to a missing folder which was created by older versions.

### How to reproduce
Steps to reproduce the behavior:
1. Play a Video with is encrypted with Widevine DRM

### Information
 - LibreELEC Version: 12.0
 - Hardware Platform: RPi4

### Log file

> 2024-04-18 14:51:55.086 T:787      info <general>: VideoPlayer::OpenFile: http://127.0.0.1:6969/manifest?path=%2Flive%2Fdisk1%2FNL_000004_019461%2Fgo-dash-fhd-hevc-pois%2FNL_000004_019461.mpd&token=YXNzVHlwPU9yaW9uLURBU0gtSEVWQyZjLWlwLXVubG9ja2VkPTEmY29uSWQ9TkxfMDAwMDA0XzAxOTQ2MSZjb25UeXBlPTQmY3VzSWQ9MTEwNzAzMjRfbmwmZGV2RmFtPXdlYi1kZXNrdG9wJmRyPTAmZHJtQ29uSWQ9bmxfdHZfc3RhbmRhYXJkX2NlbmMmZXhwaXJ5PTE3MTM0NDQ4NjQmZm49c2hhMjU2JnBhdGhVUkk9JTJGbGl2ZSUyRmRpc2sxJTJGTkxfMDAwMDA0XzAxOTQ2MSUyRmdvLWRhc2gtZmhkLWhldmMtcG9pcyUyRiUyQSZwcm9maWxlPTI1YWYyYjVjLTI5NzAtNGMwNy1iNjI2LTE3NWZiMDRmODhlNiZyZXVzZT0tMSZzTGltPTMmc2VzSWQ9ZmxlamY1RG1JaXdnYUE5RG1JaVVCd2h1SWU0MGpqVE1UZHNJJnNlc1RpbWU9MTcxMzQ0NDcxNCZzdHJMaW09MywxMDQ5ODIxNTJjOWI5YWRiODQzZjE3NDQ5YWZmMDI2NDA1MWE0Y2U4YTFhMjQzNTQwNDljYzk3NTBhMDljYWY1&hostname=wp4-vxtoken-anp-g05060506-hzn-nl.t1.prd.dyncdn.dmdsdp.com
> 2024-04-18 14:51:55.087 T:878      info <general>: Creating InputStream
> 2024-04-18 14:51:55.091 T:878   warning <general>: AddOnLog: inputstream.adaptive: Warning "inputstream.adaptive.manifest_type" property is deprecated and will be removed next Kodi version, the manifest type is now automatically detected.
>                                                    If you are using a proxy remember to add the appropriate "content-type" header to the HTTP manifest response
>                                                    See Wiki page "How to provide custom manifest/license" to learn more about it.
> 2024-04-18 14:51:55.096 T:878      info <general>: AddOnLog: inputstream.adaptive: [Repr. chooser] Resolution set: 1920x1080, max allowed: 1920x1080, Adjust refresh rate: 0
> 2024-04-18 14:51:56.311 T:878      info <general>: AddOnLog: inputstream.adaptive: Manifest successfully parsed (Periods: 1, Streams in first period: 4, Type: live)
> 2024-04-18 14:51:58.581 T:878     error <general>: AddOnLog: inputstream.adaptive: Write: Cannot create directory: /storage/.kodi/userdata/addon_data/inputstream.adaptive/widevine/5c016e8f0f95f039102cbe8366c5c7f3/
> 2024-04-18 14:51:58.611 T:878     error <general>: AddOnLog: inputstream.adaptive: SendSessionMessage: License update not successful (no keys)
> 2024-04-18 14:51:58.612 T:878     error <general>: AddOnLog: inputstream.adaptive: Initialize failed (SingleSampleDecrypter)
> 2024-04-18 14:51:58.612 T:878     error <general>: AddOnLog: inputstream.adaptive: Instances: 1
> 2024-04-18 14:51:58.615 T:878     error <general>: CVideoPlayer::OpenInputStream - error opening [http://127.0.0.1:6969/manifest?path=%2Flive%2Fdisk1%2FNL_000004_019461%2Fgo-dash-fhd-hevc-pois%2FNL_000004_019461.mpd&token=YXNzVHlwPU9yaW9uLURBU0gtSEVWQyZjLWlwLXVubG9ja2VkPTEmY29uSWQ9TkxfMDAwMDA0XzAxOTQ2MSZjb25UeXBlPTQmY3VzSWQ9MTEwNzAzMjRfbmwmZGV2RmFtPXdlYi1kZXNrdG9wJmRyPTAmZHJtQ29uSWQ9bmxfdHZfc3RhbmRhYXJkX2NlbmMmZXhwaXJ5PTE3MTM0NDQ4NjQmZm49c2hhMjU2JnBhdGhVUkk9JTJGbGl2ZSUyRmRpc2sxJTJGTkxfMDAwMDA0XzAxOTQ2MSUyRmdvLWRhc2gtZmhkLWhldmMtcG9pcyUyRiUyQSZwcm9maWxlPTI1YWYyYjVjLTI5NzAtNGMwNy1iNjI2LTE3NWZiMDRmODhlNiZyZXVzZT0tMSZzTGltPTMmc2VzSWQ9ZmxlamY1RG1JaXdnYUE5RG1JaVVCd2h1SWU0MGpqVE1UZHNJJnNlc1RpbWU9MTcxMzQ0NDcxNCZzdHJMaW09MywxMDQ5ODIxNTJjOWI5YWRiODQzZjE3NDQ5YWZmMDI2NDA1MWE0Y2U4YTFhMjQzNTQwNDljYzk3NTBhMDljYWY1&hostname=wp4-vxtoken-anp-g05060506-hzn-nl.t1.prd.dyncdn.dmdsdp.com]
> 2024-04-18 14:51:58.615 T:878      info <general>: CVideoPlayer::OnExit()
> 

### Context
I have created an addon that plays live TV streams. I have tested this on Windows 11 and it works fine. Also the stable Libreelec version (11.0) is working fine.
Here a summary of the versions I tested:

> LibreElec stable (2024.01, LibreELEC 11.0)
> 	inputstream.adaptive v20.3.18.1 => OK
> 	
> LibreElec beta (2024.04, LibreELEC 12.0)
> 	inputstream.adaptive v21.4.5.1 => inputstream.adaptive: Write: Cannot create directory: /storage/.kodi/userdata/addon_data/inputstream.adaptive/widevine/5c016e8f0f95f039102cbe8366c5c7f3/
> 	
> Windows 11
> 	inputstream.adaptive v21.4.5	=> OK
> 

Manually creation of the folder does solve the issue.

Can you have a look at the issue?

chewitt, 2024-04-18:
Please report to inputstream.adaptive developers directly, it's not an LE issue.

closed 2024-04-18 by chewitt

tgiesela, 2024-04-24:
I reported this bug to Inputstream.adaptive developers and they created a fix for it with the request to me to test it, but I can't build a binary for LE. Can you please test the solution provided by Inpustream.adaptive developers? See [Solution](https://github.com/xbmc/inputstream.adaptive/issues/1534)

heitbaum, 2024-04-24:
> I reported this bug to Inputstream.adaptive developers and they created a fix for it with the request to me to test it, but I can't build a binary for LE. Can you please test the solution provided [...]

@tgiesela test addon built https://heitbaum.libreelec.tv/inputstream.adaptive-21.4.5.1.zip