LibreELEC/LibreELEC.tv #8848: [BUG]

dga4130-renato, 2024-05-02:
Rpi 4 
libreelec 12

Argon Fan script not working after upgrading from LE11 to LE12.
debugging the argon script says:

unable to import smbus and fails the startup service.

Traceback (most recent call last):
File "argononed.py", line 4, in <module>
import smbus

Fan starts and never stops.

Reverting to LE11 all working fine.

antonlacon, 2024-05-02:
Confirm the system-tools addon is installed and enabled on LE12.

chewitt, 2024-05-03:
I'm closing this ticket becuase LE does not provide/ship any Argon case-fan scripts in our images, so if there's a bug, it's the responsibiliy of the script author to investigate and resolve it, not ours. NB: ISTR there's a thread in the forum where someone is working on a proper Argon case add-on to support their features.

dga4130-renato, 2024-05-03:
It's the system-tools addon bug.
If you login ssh and call python up
Then import smbus you will see python loaded but smbus missing even if the smbus.so lib is there.

dga4130-renato, 2024-05-03:
> I'm closing this ticket becuase LE does not provide/ship any Argon case-fan scripts in our images, so if there's a bug, it's the responsibiliy of the script author to investigate and resolve it, not [...]

Hi, maybe I didn't explain well.
Can u read the further comment ?

dga4130-renato, 2024-05-03:
> Confirm the system-tools addon is installed and enabled on LE12.

System-tools installed but smbus module not working

antonlacon, 2024-05-03:
@heitbaum this looks related to work on #7602 If I add `PYTHONPATH=/storage/.kodi/addons/virtual.system-tools/lib.private`  when calling python, it's able to import smbus.

HiassofT, 2024-05-03:
ah, good catch, the python stuff shouldn't have gone to lib.private but remained in lib (libi2c can stay in lib.private, rpath of smbus.so needs to be adjusted to $ORIGIN/../lib.private)

dga4130-renato, 2024-05-03:
> ah, good catch, the python stuff shouldn't have gone to lib.private but remained in lib (libi2c can stay in lib.private, rpath of smbus.so needs to be adjusted to $ORIGIN/../lib.private)

Should be a quick fix done in system-tools ?
If could help I can test it fast...

heitbaum, 2024-05-03:
Just testing the fix

closed 2024-05-03 by chewitt

dga4130-renato, 2024-05-03:
> Just testing the fix

Safe to install ? i mean if upgrade to le12 this fix will be present now ?

antonlacon, 2024-05-03:
Wait for tomorrow. Builders and mirrors need to get the changes.

dga4130-renato, 2024-05-04:
Installed le13 nightly where smbus should be fixed but seems the same problem as le12 release

Libreelec:~ # python
Python 3.11.9 (main, May  4 2024, 12:08:04) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import smbus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'smbus'
>>>

heitbaum, 2024-05-05:
> Installed le13 nightly where smbus should be fixed but seems the same problem as le12 release
> 
> Libreelec:~ # python Python 3.11.9 (main, May 4 2024, 12:08:04) [GCC 13.2.0] on linux Type "help", "copyright", "credits" or "license" for more information.
> 
> > > > import smbus
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > ModuleNotFoundError: No module named 'smbus'

Updated addons for LE13 have not been built yet

PR #8849 (2024-05-03): system-tools: move smbus.so to lib and addon (1)
- Fixes #8848