LibreELEC/LibreELEC.tv #8853: [BUG] missing realtek bluetooth firmware rtl8761bu

HidingCherry, 2024-05-03:
### Describe the bug
Title says it: bluetooth firmware is missing for rtl8761bu

### How to reproduce
Steps to reproduce the behavior:
1. just put in the bluetooth usb stick

### Information
 - LibreELEC Version: 12.0.0
 - Hardware Platform: odroidc2

### Log file
May 03 13:44:22 LibreELEC kernel: usb 1-1.4: new full-speed USB device number 6 using dwc2
May 03 13:44:22 LibreELEC kernel: usb 1-1.4: New USB device found, idVendor=0bda, idProduct=a729, bcdDevice= 2.00
May 03 13:44:22 LibreELEC kernel: usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 03 13:44:22 LibreELEC kernel: usb 1-1.4: Product: Bluetooth Radio
May 03 13:44:22 LibreELEC kernel: usb 1-1.4: Manufacturer: Realtek
May 03 13:44:22 LibreELEC kernel: usb 1-1.4: SerialNumber: 00E04C239987
May 03 13:44:22 LibreELEC kernel: Bluetooth: Core ver 2.22
May 03 13:44:22 LibreELEC kernel: NET: Registered PF_BLUETOOTH protocol family
May 03 13:44:22 LibreELEC kernel: Bluetooth: HCI device and connection manager initialized
May 03 13:44:22 LibreELEC kernel: Bluetooth: HCI socket layer initialized
May 03 13:44:22 LibreELEC kernel: Bluetooth: L2CAP socket layer initialized
May 03 13:44:22 LibreELEC kernel: Bluetooth: SCO socket layer initialized
May 03 13:44:22 LibreELEC kernel: usbcore: registered new interface driver btusb
May 03 13:44:22 LibreELEC kernel: Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
May 03 13:44:22 LibreELEC kernel: Bluetooth: hci0: RTL: rom_version status=0 version=1
May 03 13:44:22 LibreELEC kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
May 03 13:44:22 LibreELEC kernel: bluetooth hci0: Direct firmware load for rtl_bt/rtl8761bu_fw.bin failed with error -2
May 03 13:44:22 LibreELEC kernel: Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761bu_fw not found

### Context
firmware is available in the linux-firmware main repo, please include them:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtl_bt

HidingCherry, 2024-05-03:
fyi - on my laptop:

```
Mai 03 15:34:58 laptop kernel: usb 1-1.3: new full-speed USB device number 69 using xhci_hcd
Mai 03 15:34:58 laptop kernel: usb 1-1.3: New USB device found, idVendor=0bda, idProduct=a729, bcdDevice= 2.00
Mai 03 15:34:58 laptop kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mai 03 15:34:58 laptop kernel: usb 1-1.3: Product: Bluetooth Radio
Mai 03 15:34:58 laptop kernel: usb 1-1.3: Manufacturer: Realtek
Mai 03 15:34:58 laptop kernel: usb 1-1.3: SerialNumber: 00E04C239987
Mai 03 15:34:58 laptop kernel: Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
Mai 03 15:34:58 laptop kernel: Bluetooth: hci1: RTL: rom_version status=0 version=1
Mai 03 15:34:58 laptop kernel: Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
Mai 03 15:34:58 laptop kernel: Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_config.bin
Mai 03 15:34:58 laptop kernel: Bluetooth: hci1: RTL: cfg_sz 6, total sz 30210
Mai 03 15:34:58 laptop kernel: Bluetooth: hci1: RTL: fw version 0xdfc6d922
Mai 03 15:34:59 laptop kernel: Bluetooth: MGMT ver 1.22

```

chewitt, 2024-05-03:
PR #8854 will resolve it for future LE12 releases. Until then you can do this: https://wiki.libreelec.tv/how-to/add-firmware

HidingCherry, 2024-05-03:
@chewitt fyi - this is an external bluetooth usb dongle, it is not part of the odroid c2 board (which has neither bluetooth, nor wifi on board).
Because it is missing, I decided to buy that dongle and noticed that not every linux-firmware image for bluetooth devices are included.

They are not big (for realtek), I would've guessed you would include them by default.

chewitt, 2024-05-03:
LE has a long history of cherry-picking the firmware to include based on actual user demand/need rather than trying to guess in advance what clumps of the 350MB+ linux-firmware repo might be required. Adding the bits to AMLGX isn't an issue if they'll be used.

heitbaum, 2024-05-25:
Closing now pr is merged.

PR #8854 (2024-05-24): amlogic: include rtl8761bu firmware
Pick the rtl8176bu firmware for future AMLGX LE12 images. Fixes #8853. I'll submit the same change to master along with a kernel bump in the near future.
closed 2024-05-25 by heitbaum