LibreELEC/LibreELEC.tv #8857: [BUG] reset SuperSpeed USB device number x using xhci-hcd

pipsen, 2024-05-04:
### Describe the bug
Setup:
- Raspberry Pi 5 with original power supply
- 64GB microSD card SanDisk Extreme Pro
- LibreELEC 12.0.0 release image
- tested with two different wifi USB dongles
  * TP-Link Archer T3U AC1300
  * TP-Link Archer T3U Plus AC1300
- tested with two different USB 3.0 external 2,5'' harddrives

### How to reproduce
Steps to reproduce the behavior:
1. Boot up LibreELEC with connected USB disk and USB WIFI dongle
2. copy larger files from internal microSD card to external USB drive

Result after a short period of time: WIFI dongle shutdown - no connection any more
WIFI connection not visible any more in LibreELEC configuration GUI
no ping possible any more

### Information
 - LibreELEC Version: 12.0.0
 - Hardware Platform: RPi 5 8GB

### Log file
[dmesg.log](https://github.com/LibreELEC/LibreELEC.tv/files/15209713/dmesg.log)

relevant part:
`[ 1322.378323] usb 2-1: reset SuperSpeed USB device number 2 using xhci-hcd
[ 1322.393578] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
[ 1322.393583] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 2a 4e 00 00 08 00 00
[ 1322.393585] I/O error, dev sda, sector 2772480 op 0x0:(READ) flags 0x80700 phys_seg 34 prio class 2
[ 1322.516329] usb 2-1: reset SuperSpeed USB device number 2 using xhci-hcd`

### Context
n/A

heitbaum, 2024-10-29:
Is this still occurring on the nightly LE12/LE13 images? Suggest raising this on the forum for assistance,

heitbaum, 2024-11-02:
If still an issue. Please raise on the forums.

closed 2024-11-02 by heitbaum