LibreELEC/LibreELEC.tv #8858: [BUG] python not import modules

dga4130-renato, 2024-05-04:
Rpi4 
Installed le13 04.05.2024 nightly where smbus python module should be fixed but seems the same problem as le12 release

Libreelec:~ # python
Python 3.11.9 (main, May 4 2024, 12:08:04) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.

            import smbus
            Traceback (most recent call last):
            File "", line 1, in
            ModuleNotFoundError: No module named 'smbus'

If use export pythonpath option to reach private.lib path you can import smbus.so but  not RPiO.GPIO .

Quit a problem because you can't control fan voltage without these modules.

heitbaum, 2024-05-05:
Hi @dga4130-renato the fix is in the updated system-tools -12.0.0.1 (not the nightly image) please check your addon version. The stubs.so will be in lib (not lib.private)

heitbaum, 2024-05-05:
- duplicate of #8848

closed 2024-05-05 by heitbaum

dga4130-renato, 2024-05-05:
> Hi @dga4130-renato the fix is in the updated system-tools -12.0.0.1 (not the nightly image) please check your addon version. The smbus.so will be in lib (not lib.private)

Ops. My fault. Smbus loading ok but rpi.gpio missing ... Unable to load module . Maybe same problem with system-tools ?

heitbaum, 2024-05-05:
rpi.gpio has been replaced in LE12 -
https://github.com/LibreELEC/LibreELEC.tv/pull/8230