LibreELEC/LibreELEC.tv #8864: [BUG]Partition creation for storage is tiny

swanie98635, 2024-05-06:
Use imager to create an image on the newly partitioned and formatted NVME drive.
Boots fine - but then the storage partition is like 128mb - too small for even one media file
Tried using Gparted from SD card Raspian to grow - can't get it to grow the partition

### Information
 - LibreELEC Version: 12.0.0
 - Hardware Platform: Rpi5 4 gb with Pimoroni NVME hat (crucial nvme board working fine)

### Log file

### Context
Can't seem to get the storage partition to correctly size, either when reinstalling (which I've done 3x) or by gparted (which I tried a few times as well.  The NVME board is fine.  I've actually got it formatted now to the 3.6 GB formatted size as a partition.  I can access it no problem.  Just seems LibreElec does not like large NVME devices.

mglae, 2024-05-08:
Wild guess (because any technical information is missing): writing the 512 bytes/sector LE image to a 4096 bytes/sector NVME can only fail.

heitbaum, 2024-05-25:
No technical detail/logs/or update from reporter. Closing ticket. Please reopen or use forum if further information is available.

closed 2024-05-25 by heitbaum