LibreELEC/LibreELEC.tv #8884: [BUG] LibreELEC Backup + Docker Add-On

JerryCrazy, 2024-05-09:
Hello LibreELEC team

If the Docker add-on is installed, I can make a "Backup" from the Kodi interface,
It works so far, but if I try to decompress the TAR archive to check that the backup is correct in my computer, I get an archive decompression error.

tar: storage/.kodi/userdata/addon_data/service.system.docker/docker/volumes/backingFsBlockDev : Cannot mknod: Operation not permitted

Because of this, I don't trust my backup.

Can we put exceptions on the files to not back them up?

I use LibreELEC 12.0.0 in Raspberry Pi 4, but previous version of OS have also same problem with Docker add-on.

mglae, 2024-05-12:
LE backup is created as root and has to be extracted as root.

Normal users do not have the permissions to create device nodes.

heitbaum, 2024-05-12:
Not a bug. Please use forum for general help and support.

closed 2024-05-12 by heitbaum