LibreELEC/LibreELEC.tv #8888: [BUG] kodi-bin error loading libspdlog.so.1.13

nvdias0, 2024-05-13:
### Describe the bug
I've just compiled devel-20240513103038-bcd4d7f

### How to reproduce
Kodi does not load.

### Information
 - LibreELEC Version: 13 devel-20240513103038-bcd4d7f
 - Hardware Platform: RPi4

### Context
- Running:
**/usr/lib/kodi/kodi.bin**

- Returns:
**error while loading shared libraries: libspdlog.so.1.13: cannot open shared object file: No such file or directory**

- Looking into /usr/lib, I have:
/usr/lib/libspdlog.so
/usr/lib/libspdlog.so.1.14
/usr/lib/libspdlog.so.1.14.1

But not version 13.

lrusak, 2024-05-13:
Please use the forums for support.

You should do a clean build.

nvdias0, 2024-05-13:
Yep. Clean built. Gave the error. That's why I thought it was a bug on dependencies.

- So, repeated the clean:
PROJECT=RPi DEVICE=RPi4 ARCH=arm make clean

- Recompiled:
PROJECT=RPi DEVICE=RPi4 ARCH=arm make image

(Now it worked)

closed 2024-05-13 by nvdias0