LibreELEC/LibreELEC.tv #8900: [BUG] Missing Bluetooth firmware on Wetek Play 2 (brcmfmac4330)

i3laze, 2024-05-17:
### Describe the bug
Original Bluetooth module doesn't work on stock 12.0 WetekPlay2 firmware: 
https://releases.libreelec.tv/LibreELEC-AMLGX.aarch64-12.0.0-wetek-play2.img.gz

"No Bluetooth device found"

### How to reproduce
Steps to reproduce the behavior:
1. Go to 'Kodi Settings - LibreELEC - Services'.
2. Verify Bluetooth service is Enabled.
3. Go to 'Kodi Settings - LibreELEC - Bluetooth.'
3. See error "No Bluetooth device found"

### Information
 - LibreELEC Version: 12.0
 - Hardware Platform: WeTek Play 2, flashed to eMMC:
 https://wiki.libreelec.tv/hardware/amlogic/wetek-hub-play2
 
### Log file
PasteKodi: https://paste.libreelec.tv/smooth-cod.log
dmesg output:  https://paste.libreelec.tv/prompt-grouper.log

### Context
Bluetooth original module inside the unit was working fine on latest CoreELEC 9.2.8 before LE12 clean install.

i3laze, 2024-05-17:
While issue seems a clone of Realtek firmware missing https://github.com/LibreELEC/LibreELEC.tv/issues/8853

It goes without saying you expect native Bluetooth dongle to be supported by native Wetek Play 2 image.

heitbaum, 2024-05-17:
Please test following https://github.com/LibreELEC/LibreELEC.tv/issues/8853#issuecomment-2093339726 if it works then this will be closed with #8853

chewitt, 2024-05-17:
LE includes the firmware and the device/driver is probed and working from the console:

```WP2:~ # dmesg | grep -i blue
[    9.922582] Bluetooth: Core ver 2.22
[    9.923065] NET: Registered PF_BLUETOOTH protocol family
[    9.923132] Bluetooth: HCI device and connection manager initialized
[    9.923804] Bluetooth: HCI socket layer initialized
[    9.923925] Bluetooth: L2CAP socket layer initialized
[    9.924676] Bluetooth: SCO socket layer initialized
[   10.087105] Bluetooth: HCI UART driver ver 2.3
[   10.087145] Bluetooth: HCI UART protocol H4 registered
[   10.087354] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   10.087795] Bluetooth: HCI UART protocol Broadcom registered
[   10.087900] Bluetooth: HCI UART protocol QCA registered
[   10.469738] Bluetooth: hci0: BCM: chip id 62
[   10.472064] Bluetooth: hci0: BCM: features 0x0f
[   10.495191] Bluetooth: hci0: BCM4330B1
[   10.495236] Bluetooth: hci0: BCM4330B1 (002.001.003) build 0000
[   10.522116] Bluetooth: hci0: BCM4330B1 'brcm/BCM4330B1.hcd' Patch
[   16.114702] Bluetooth: hci0: BCM: features 0x0f
[   16.137743] Bluetooth: hci0: Win8 Tablet Semco SWB-B42G - BCM4330B1 37.4 MHz Class 2 NoExtLNA
[   16.137764] Bluetooth: hci0: BCM4330B1 (002.001.003) build 1316
[   16.250556] Bluetooth: MGMT ver 1.22
WP2:~ # bluetoothctl show
Controller 43:30:A0:00:1F:AC (public)
	Manufacturer: 0x000f (15)
	Version: 0x06 (6)
	Name: WP2
	Alias: WP2
	Class: 0x000c0000 (786432)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d054B
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x05 (5)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
```
and the card is active:
```
WP2:~ # btmon 
Bluetooth monitor ver 5.75
btmon[2652]: = Note: Linux version 6.9.0 (aarch64)                                                                                                           0.931248
btmon[2652]: = Note: Bluetooth subsystem version 2.22                                                                                                        0.931261
= New Index: 43:30:A0:00:1F:AC (Primary,UART,hci0)                                                                                                    [hci0] 0.931267
= Open Index: 43:30:A0:00:1F:AC                                                                                                                       [hci0] 0.931269
= Index Info: 43:30:A0:00:1F:AC (Broadcom Corporation)                                                                                                [hci0] 0.931274
bluetoothd[527]: @ MGMT Open: bluetoothd (privileged) version 1.22                                                                                  {0x0001} 0.931280
> HCI Event: Inquiry Complete (0x01) plen 1                                                                                                        #1 [hci0] 4.763176
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                                                                    {0x0001} [hci0] 4.763271
        Address type: 0x07
          BR/EDR
          LE Public
[... 19 lines cut ...]
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                            #6 [hci0] 10.327426
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                       #7 [hci0] 10.329142
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                                                              {0x0001} [hci0] 10.329366
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                                                                                                   {0x0001} [hci0] 10.329418
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Enabled (0x01)
```
However LE settings does indeed show "No Bluetooth device found." so I'm wondering if we have an issue with the d-bus agent or however else the device is detected?

TL/DR: the workaround will be to pair/connect things using `bluetoothctl` over SSH not the settings add-on.

i3laze, 2024-05-17:
Thank you for checking.  Will try the workaround via SSH.
Luckily @chewitt has his old Play 2 unit : )

Let this issue stay open until resolved.

i3laze, 2024-09-17:
Out of curiosity I got TP-Link UB500 USB dongle connected.
On LE12 it seems to work without extra drivers, yet Play2 add-on still shows “no device found”.
LE v12.0.1 included.

`LibreELEC:~ # dmesg | grep -i blue
[    9.030297] Bluetooth: Core ver 2.22
[    9.030477] NET: Registered PF_BLUETOOTH protocol family
[    9.030488] Bluetooth: HCI device and connection manager initialized
[    9.030899] Bluetooth: HCI socket layer initialized
[    9.030921] Bluetooth: L2CAP socket layer initialized
[    9.031336] Bluetooth: SCO socket layer initialized
[    9.307407] Bluetooth: HCI UART driver ver 2.3
[    9.307440] Bluetooth: HCI UART protocol H4 registered
[    9.307623] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.308192] Bluetooth: HCI UART protocol Broadcom registered
[    9.308299] Bluetooth: HCI UART protocol QCA registered
[    9.327811] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    9.329778] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    9.329817] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[    9.358682] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[    9.362264] Bluetooth: hci0: RTL: cfg_sz 6, total sz 30210
[    9.683973] Bluetooth: hci1: BCM: chip id 62
[    9.686212] Bluetooth: hci1: BCM: features 0x0f
[    9.709339] Bluetooth: hci1: BCM4330B1
[    9.709372] Bluetooth: hci1: BCM4330B1 (002.001.003) build 0000
[    9.738692] Bluetooth: hci1: BCM4330B1 'brcm/BCM4330B1.hcd' Patch
[   11.485883] Bluetooth: hci0: command 0xfc20 tx timeout
[   11.485922] Bluetooth: hci0: RTL: Failed to generate devcoredump
[   15.505465] Bluetooth: hci1: BCM: features 0x0f
[   15.528556] Bluetooth: hci1: Win8 Tablet Semco SWB-B42G - BCM4330B1 37.4 MHz Class 2 NoExtLNA
[   15.528588] Bluetooth: hci1: BCM4330B1 (002.001.003) build 1316
[   15.642893] Bluetooth: MGMT ver 1.22
[   18.503541] Bluetooth: RFCOMM TTY layer initialized
[   18.503605] Bluetooth: RFCOMM socket layer initialized
[   18.503650] Bluetooth: RFCOMM ver 1.11
[   19.676974] Bluetooth: hci0: RTL: download fw command failed (-110)
[ 2312.921254] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[ 2312.923118] Bluetooth: hci0: RTL: rom_version status=0 version=1
[ 2312.923156] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[ 2312.924295] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[ 2312.924692] Bluetooth: hci0: RTL: cfg_sz 6, total sz 30210
[ 2313.198031] Bluetooth: hci0: RTL: fw version 0xdfc6d922
[ 2313.323947] Bluetooth: MGMT ver 1.22 

LibreELEC:~ # btmon
Bluetooth monitor ver 5.75
btmon[2015]: = Note: Linux version 6.8.0 (aarch64)                                                                 0.018263
btmon[2015]: = Note: Bluetooth subsystem version 2.22                                                              0.018282
= New Index: E8:48:B8:C8:20:00 (Primary,USB,hci0)                                                           [hci0] 0.018292
= Open Index: E8:48:B8:C8:20:00                                                                             [hci0] 0.018298
= Index Info: E8:48:B8:C8:20:00 (Realtek Semiconductor Corporation)                                         [hci0] 0.018305
= New Index: 43:30:A0:00:1F:AC (Primary,UART,hci1)                                                          [hci1] 0.018314
= Open Index: 43:30:A0:00:1F:AC                                                                             [hci1] 0.018320
= Index Info: 43:30:A0:00:1F:AC (Broadcom Corporation)                                                      [hci1] 0.018325
bluetoothd[635]: @ MGMT Open: bluetoothd (privileged) version 1.22                                        {0x0001} 0.018336
> HCI Event: Inquiry Complete (0x01) plen 1                                                              #1 [hci1] 3.773533
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                          {0x0001} [hci1] 3.773737
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Disabled (0x00)
LibreELEC:~ # bluetoothctl show
Controller E8:48:B8:C8:20:00 (public)
        Manufacturer: 0x005d (93)
        Version: 0x0a (10)
        Name: LibreELEC #1
        Alias: LibreELEC
        Class: 0x000c0000 (786432)
        Powered: yes
        PowerState: on
        Discoverable: no
        DiscoverableTimeout: 0x000000b4 (180)
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d054B
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x04 (4)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
        SupportedSecondaryChannels: 1M
        SupportedSecondaryChannels: 2M
        SupportedSecondaryChannels: Coded `

heitbaum, 2024-11-04:
Please confirm if this is still occurring with the latest LE13 nightlies?

chewitt, 2024-11-04:
Yes, the BT device is probed and drivers loaded and all looks good, but btmon shows zero traffic on the BT interface and you’d normally see constant chatter from other devices nearby.

i3laze, 2026-01-16:
I'll be able to confirm it in next 15 days(