LibreELEC/LibreELEC.tv #8902: [BUG] Stereo upmix is broken

ojaksch, 2024-05-18:
### Describe the bug
When enabling _Stereo Upmix_ and using a multichannel receiver (5.1 here), sound on one or more speakers is missing

### How to reproduce
Steps to reproduce the behavior:
1. Go to Settings/System/Audio
2. Enable _Stereo Upmix_
3. Play a stereo sound file

### Information
 - LibreELEC Version: 12.0.0
 - Hardware Platform: Generic x86_64

### Log file
None added. Bug is known, present and fixed in upstream, see [here](https://github.com/xbmc/xbmc/issues/25028)

### Context
Would be glad to see this addressed in a nightly, if not already done

i3laze, 2024-05-18:
Never tried that feature myself,
Sorry for critics, but I would recommend to do upmix 2.0 to 5.1 using proprietary receiver logic, rather than software. Or just enjoy native stereo scene.

ojaksch, 2024-05-18:
Your criticism is completely fine. I **did** upmixing by receiver in the past, this is working but sound is flat then. That's why I'm using upmixing by Kodi for years with great success and happy ears :) And it's a bug. We don't like bugs.

heitbaum, 2024-05-18:
Confirmed merged and included in Omega 21.1 milestone.

DDQW, 2024-06-28:
will this include activating the subwoofer in a 2.1 setup? i am on 13 beta and my sub does jack shit when playing stereo sources

chewitt, 2024-06-28:
This might be relevant https://github.com/xbmc/xbmc/pull/25395

fritsch, 2024-06-28:
Not relevant. The linked PR allows the lfe channel being distributed to LR and LF when down mixing a 5.1 source. By default the LFE is left out, while it makes sense to have it on full range speakers.

For a 2.1 setup to cause some bass even on stereo input the AVR should have its own setting. It is not forseen in Kodi as doing this with sense would require a frequency filter. Shuffling FL and FR into the sub 1:1 would end up in something not so funny: over bass.

DDQW, 2024-06-29:
that is kinda what i am doing right now. i connect the sub directly to fl and fr output then it does its own bass filtering (fixed at 80hz) and gives the rest to the speaker.

The downside is i have switch cables when i want to watch something that does have lfe.

ojaksch, 2024-08-27:
LE 12.0.1 is released, my box updated and stereo upmix is working again. Yikes, finally decent "stereo" sound again on all speakers!
Thanks for listening, fixing and any hint I got here.

closed 2024-08-27 by ojaksch