LibreELEC/LibreELEC.tv #8941: [BUG] Kernel (6.9.2)  crash/warning

eldersixpack13, 2024-05-30:
### Describe the bug

a kernel crash/warning in dmesg
LE/Kodi runs normal though

### How to reproduce
Steps to reproduce the behavior:
1. install LibreELEC-Generic.x86_64-13.0-nightly-20240529-c8e0f92.img.gz
2. reboot
3. See dmesg

### Information
 - LibreELEC Version: LibreELEC-Generic.x86_64-13.0-nightly-20240529-c8e0f92
 - Hardware Platform: NUC13
 - dmesg attached 
 see timestamp:  2.553550 or line 908

### Log file
[dmesg_2024-05-30.txt](https://github.com/LibreELEC/LibreELEC.tv/files/15491777/dmesg_2024-05-30.txt)

### Context
I do not see this bug/crash/warning on my Desktop (11th Gen Intel i5-11400) through the whole 6.9. x Kernel serie

no kodi.log attached cause nothing related in

eldersixpack13, 2024-05-30:
a short search revealed this (but it's from march !):

https://marc.info/?l=linux-kbuild&m=171148507323465&w=4

heitbaum, 2024-07-15:
Known issue with the broadcom OOT wl driver. There are no fixes available for this at this time.
  - [broadcom-wl-dkms](https://gitlab.archlinux.org/archlinux/packaging/packages/broadcom-wl-dkms/-/commits/main)
    - Causing thunk warning
    - https://github.com/torvalds/linux/commit/4461438a8405e800f90e0e40409e5f3d07eed381
    - https://bbs.archlinux.org/viewtopic.php?id=296729
    - https://discussion.fedoraproject.org/t/kernel-hangs-since-some-recent-fedora-kernels/120846/6?replies_to_post_number=5
    - https://bugzilla.rpmfusion.org/show_bug.cgi?id=6991
    - https://discussion.fedoraproject.org/t/kernel-hangs-since-some-recent-fedora-kernels/120846/17

We will include a patch when it is made available.

heitbaum, 2024-09-08:
Tested with changes from akmod-wl unsuccessfully.
 
- https://koji.rpmfusion.org/kojifiles/packages/wl-kmod/6.30.223.271/53.fc40/src/wl-kmod-6.30.223.271-53.fc40.src.rpm
- https://bugzilla.rpmfusion.org/show_bug.cgi?id=6991

Given that this has yet to be fixed upstream it is likely that the wl (bcm_sta) wireless driver will have to be dropped from the LE13 release.

The target kernel for LE13 is likely to be the 6.12 LTS kernel - which probably starts its merge window 16th September 2024.

eldersixpack13, 2024-09-08:
fine by me.
I'm running a nuc with Intel Wifi.

feel free to close this bug

PR #9522 (2024-11-23): bcm_sta: drop wireless card driver
This driver has not been working since kernel 6.10
- no fixes have been forthcoming
- closes #8941
closed 2024-11-23 by chewitt