LibreELEC/LibreELEC.tv #9003: [BUG] cec broken

damnms, 2024-06-18:
### Describe the bug

I installed LE12 on a rpi5. I noticed that it does not power on/power off when i power on/power off my TV. Unfortunately this is a killer feature for me, so i switched back to an amlogic device with coreelec. All works here, so its not related to the other devices.

### How to reproduce
Steps to reproduce the behavior:
1. power off the TV
2. the rpi5 stays on

### Information
 - LibreELEC Version: 12
 - Hardware Platform: rpi5

### Log file

### Context

https://forum.libreelec.tv/thread/28640-cec-problems-on-rpi5-with-le-nightly/

henning-schild, 2024-09-05:
seems like a duplicate of #9002

damnms, 2024-09-05:
#9002 is about a not saved configuration. #9003 is that powering off/on the device does just nothing. I am not sure if they are related.

heitbaum, 2024-09-15:
- Closing now that #9254 and #9255 are merged in nightlies. If still an issue please raise on the forum/a bug can be raised.

closed 2024-09-15 by heitbaum