LibreELEC/LibreELEC.tv #9005: [BUG] Meteor Lake w/Intel ARC Installer Freezes on Boot Mode Selection

xags, 2024-06-19:
### Describe the bug
When using a properly created USB installer image, the USB stick boots to the install screen waiting to select a boot mode (installer,live,run). Once selecting any of the boot modes the system freezes.

### How to reproduce
Steps to reproduce the behavior:
1. Boot via USB (UEFI or Legacy)
2. Install selection screen shows
3. Select either installer, live, or run.
4. Machine freezes, USB appears to disconnect both the keyboard and USB stick.

### Information
 - LibreELEC Version: 12.0 and latest 6/19/24 13.0 nightly
 - Hardware Platform: Asus NUC Pro ( NUC14RVK with Intel Core Ultra 5 125H Processor)

### Log file
No messages are logged by the installer itself.

### Context
Unable to get the installer to even start.

xags, 2024-06-19:
I've also tried modifying the syslinux.cfg on the USB stick to remove the quiet option. The only additional output is: "Loading kernel: OK"

smp79, 2024-06-19:
I think CONFIG_PINCTRL_METEORLAKE needs to be enabled in kernel config.

smp79, 2024-06-20:
Could you try [this image](https://www.dropbox.com/scl/fi/g35qlpbv8lgnskpiqr4h3/LibreELEC-Generic.x86_64-13.0-devel-20240620031939-fb39b80.img.gz?rlkey=i87qbod97f2b4b217x7ugpkcx&st=euoi2wjz&dl=1)?

xags, 2024-06-20:
I'm not sure what I did but the 6/19/24 nightly started working after I disabled VT-D/IVT/TPM and maybe CEC. What's frustrating is that when I re-enable them the 13.0 nightly still boots properly!  If I default the bios to "optimal defaults" it fails until I repeat the enable/disable procedure. Guessing a BIOS bug?

Your image doesn't cause the machine to lock-up but it gives an error: Error in mount_flash: mount_common: Could not mount UUID=2006-0908

The official 12.0 image does the same thing as your image with a different UUID in the error message. I've dd'd the image multiple times and re-downloaded to ensure it wasn't corrupt.

illashiza, 2024-07-01:
The LibreELEC USB installer freezes on boot mode selection with Asus NUC Pro (NUC14RVK). Both UEFI and Legacy modes result in keyboard and USB disconnection.

heitbaum, 2024-09-08:
Fixed and merged in LE13
- https://forum.libreelec.tv/thread/28963-x86-64-le-12-0-1-on-intel-meteor-lake/
- #9267

closed 2024-09-08 by heitbaum