LibreELEC/LibreELEC.tv #9023: [BUG]

THEJAILBRAKEMAN, 2024-06-25:
### Describe the bug
Good morning Devs !. I updated my phone to IOS 18 yesterday, am airplay doesn't work anymore. I tried another airplay receiver (ShairPlay-Sync) and it's working totally fine. and it using Airplay 2. I tried to reset all the settings but it's still not working. 

### How to reproduce
Steps to reproduce the behavior:
Have a device on IOS 18 Dev Build
Try to connect to the device 
You will get the following error, Unable to connect to "LivingRoomTV"(Custom Name) 

### Information
 - LibreELEC Version: LibreELEC (officiel ) :12.0.0 (Kernel : Linux 6.6.28)
 - Hardware Platform: Generic AMD thin Client. HP t520

### Log file
https://paste.libreelec.tv/relaxed-baboon.log

### Context
Tying to connect to Airplay Using IOS18

chewitt, 2024-06-25:
No issues playing from an iPhone on v17.5.1 to an LE13 (same version of Kodi as LE12) box here. It was presumably working before the self-inflicted update to the iOS 18 beta? 

Please provide a debug log. I doubt it will show anything more .. but you never know.

heitbaum, 2024-09-11:
There were a number of issues with the IOS18 beta and AirPlay1. Some are discussed on the https://github.com/mikebrady/shairport-sync

With IOS18 (Released version) and using https://github.com/juhovh/shairplay which is used in LE12/13 Kodi Omega and Piers is generating the following TCP Stream (and then erroring on the iPhone.)
```
OPTIONS * RTSP/1.0
CSeq: 0
DACP-ID: 36D06E75816F32C3
Active-Remote: 458539719
User-Agent: AirPlay/800.74.1

RTSP/1.0 200 OK
CSeq: 0
Apple-Jack-Status: connected; type=analog
Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER

OPTIONS * RTSP/1.0
CSeq: 1
DACP-ID: 36D06E75816F32C3
Active-Remote: 458539719
User-Agent: AirPlay/800.74.1

RTSP/1.0 200 OK
CSeq: 1
Apple-Jack-Status: connected; type=analog
Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER

OPTIONS * RTSP/1.0
CSeq: 2
DACP-ID: 36D06E75816F32C3
Active-Remote: 458539719
User-Agent: AirPlay/800.74.1

RTSP/1.0 200 OK
CSeq: 2
Apple-Jack-Status: connected; type=analog
Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER

OPTIONS * RTSP/1.0
Apple-Challenge: 92Z+leB3H67dPd8g9/yTyw==
CSeq: 3
DACP-ID: 36D06E75816F32C3
Active-Remote: 458539719
User-Agent: AirPlay/800.74.1

RTSP/1.0 200 OK
CSeq: 3
Apple-Jack-Status: connected; type=analog
Apple-Response: t24z1XchEch8Ng7g6TZw48aFnK/wB/G09jxLCvMSs1bT0BbjhLQyog/xLTZOR0NObxPk8r7AliNMoqG0scuezYYgPjZVPwNNCgIyWBYj0iMnLZ6NaGn9YFvqIwKyO7WTMJQiqhJdwb26PiQ/xWUnXdo/8J9qdvHE6i7WL7O/PAwLARkf8ZjH/3lZ2tRUwGik2K19hB/McAXL7pUVjwYLzEObnOrxlg4K8O5MMNndMIDaB+BxTOg1TDdbcoOUB1tUo0+fKw5rwAohZVLlxwlJC3WzVHBMN4lH4l6w6IqY2uOI6ANSdxBM5khhBJyDCRS9gjFZjJSwi7l8tc5HEXPaJg
Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER
```

heitbaum, 2024-09-12:
I have retested after a reboot of the phone - both IOS17 and IOS18 are working fine. Issue probably was the beta version as discussed in the shairport-sync ticket

closed 2024-09-12 by heitbaum