LibreELEC/LibreELEC.tv #9049: [BUG] Kernel dump and freeze after upgrade to LE11/12 (rcu_sched detected stalls on CPUs/tasks)

xags, 2024-07-02:
### Describe the bug
I run five Intel NUC NUC6i7KYK (skull canyon) devices. These have been running a version of LE for well over 7-8 years. After an upgrade to either LE 11.0.6 or 12.0 I have have been experiencing random lockups on all devices.

### How to reproduce
Steps to reproduce the behavior:
1. Run LE11.0.6
2. While playing either live TV or a movie the system freezes randomly and no user input is allowed.
3. Device is unable to be rebooted via SSH or powered down via IR. The device must be unplugged or the power button physically held.
4. Downgrading to 10.x and no issues across all devices.

### Information
 - LibreELEC Version: 11.0.6/12.0
 - Hardware Platform: Intel NUC NUC6i7KYK x 5

### Log file
kodi.log (freeze somewhere between 20:46 - 24:00: https://paste.kodi.tv/xewukarafi.kodi
dmesg: https://paste.kodi.tv/aqabanenub.kodi
### Context
dmesg/journalctl rolled over and the kodi.log file appears to not show any issue. When it happens with debug logging turned on, no additional information is present in the kodi.log file.

heitbaum, 2024-07-03:
Generic LE10.0 is equivalent to Generic-legacy 11/12.
please try installing that image. 

https://libreelec.tv/downloads/generic/

if that runs stablely on your devices, then investigation could be done to see why X11 versus GBM is failing on your devices.

heitbaum, 2024-07-05:
Closing as answered. Please visit our forum for further assistance.

closed 2024-07-05 by heitbaum