LibreELEC/LibreELEC.tv #9065: [BUG] UI items, subtitles causes playing video to glitch weirdly

XxAcielxX, 2024-07-07:
### Describe the bug
When playing videos with `ff-h264-drm_prime (HW)`, `ff-vp9-drm_prime (HW)` Decorders causes video to glitch weirdly (UI items like changing volume, mute/unmute, Video Info, etc). Even the Subtitle causes this issue as well.

The issue does not occur with `ff-hevc-drm_prime (HW)` Decorder.

It is happening on a fresh test install too. Here is a video example https://youtu.be/IRxyoZ4qt_w

### How to reproduce
Steps to reproduce the behavior:
1. Play any video with h264/VP9 Codecs
2. Mute/unmute, disable/enable subtitles, open video info
3. Observe the glitch

### Information
 - LibreELEC Version: 12.0.0
 - Hardware Platform: NanoPC-T4 (RK3399)

NetDwarf, 2024-08-27:
Can confirm on other devices as well with Allwinner and RockChip SoCs.

### Problem description

* triggered by moving menu elements or multiple overlay menus or subtitles, basically anything overlaying
* i.e. triggered by opening control menu (OK/Enter) -> settings
* the video frames play out of order and often also redraw half-way (tearing)
* this happens on many, but not all hardware-accelerated videos (H264 Youtube videos consistently exhibit this problem)

### Information

* LibreELEC Version: 12.0-nightly-20240802-cbf0545
* Hardware Platform: Rock64 (RK3328)

* LibreELEC Version: 12.0.0 (stable)
* Hardware Platform: QPlus (Allwinner H6)

I also tested various versions and this has been an issue since quite a while, but I thought it'd be fixed in stable.

### Counter Example

* LibreELEC Version: 12.0.0 (stable)
* Hardware Platform: Odroid C2 (Amlogic S905)

This has various other problems, but not the flickering issue on overlay.

XxAcielxX, 2024-08-27:
I thought I was the only with this issue, but it affects many devices.

XxAcielxX, 2024-08-30:
Upgraded to LE 12.0.1 (stable), and I'm still noticing the issue happening.

knaerzche, 2024-08-30:
This has nothing to do with the kernel version, it's yet another issue of  https://github.com/xbmc/xbmc/pull/25588 (mostly because frames (-rates) of gui/video layer are out of sync). See also https://github.com/xbmc/xbmc/pull/25588#issuecomment-2278430901
@heitbaum best add https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/patches/kodi/000-temp-revert-fences.patch for RK and AW as well.

NetDwarf, 2024-09-04:
> This has nothing to do with the kernel version, it's yet another issue of [xbmc/xbmc#23921](https://github.com/xbmc/xbmc/pull/23921) (mostly because frames (-rates) of gui/video layer are out of sync). See also [xbmc/xbmc#25588 (comment)](https://github.com/xbmc/xbmc/pull/25588#issuecomment-2278430901) @heitbaum best add https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/patches/kodi/000-temp-revert-fences.patch for RK and AW as well.

Compiled a LibreELEC image with your patches and it works; no tearing. I'm honestly impressed.

### Test Setup
1. used LibreELEC 12.0.1 as base (the tag 12.0.1)
2. added https://github.com/LibreELEC/LibreELEC.tv/pull/9256 (to make it bootable)
3. copied the suggested patch over to the Rockchip folder
4. compiled

More precisely I compiled the bootable version first and then added your suggested patch to make sure I had a booting baseline.

### Allwinner also needs this patch
Just as a small reminder.

NetDwarf, 2024-09-07:
Not to make a hurry, but I am not sure about the conventions here. Am I supposed to make the PR? I am also a bit hesitant because of the authorship because this is basically @knaerzche's  fix :sweat_smile: 

Or should I just wait? (also fine, just want to cover all bases)

chewitt, 2024-09-07:
@NetDwarf .. @knaerzche or @heitbaum won't complain if someone saves them the effort of sending a pull request. Feel free to send the change suggested.

NetDwarf, 2024-09-09:
The issue is pretty much fixed by #9277 . I realized too late that the first/linked PR was only for LE13 and not LE12. Given that the issue at hand is in LE12, I would close it for now. It could potentially be reopened if there is no fix provided for LE13.

hbiyik, 2024-09-09:
Sorry to interrupt, but the fix proposed is completely disabling Async rendering,
https://github.com/xbmc/xbmc/pull/25588 here i provided a fix to prevent Async rendering messing up the rendering path in case of an unexpected drm commit, if this commit also fixes this issue, would it make sense to still include async rendering the fix https://github.com/xbmc/xbmc/pull/25588, async rendering  is really a good feature to improve performance overall.

On the other hand, the fix can be tested in wider install base like LE.

knaerzche, 2024-09-09:
> Sorry to interrupt, but the fix proposed is completely disabling Async rendering,

Sure, because it doesn't work as expected.

> https://github.com/xbmc/xbmc/pull/25588 here i provided a fix to prevent Async rendering messing up the rendering path in case of an

I checked this already, but it doesn't help with this issue here.

hbiyik, 2024-09-09:
@knaerzche ah ok, thanks for the clarifcation, i guess i misinterpreted the case here.

NetDwarf, 2024-09-11:
@XxAcielxX can you test [LE12-nightly for NanoPC-T4](https://test.libreelec.tv/12.0/Rockchip/nanopc-t4/LibreELEC-RK3399.aarch64-12.0-nightly-20240910-bd771c9-nanopc-t4.img.gz)?

If it fixes the flicker/tearing issue for you, this could be closed.

PS.: To be a bit more verbose. It was a mistake by me to PR the fix to master. I realized too late that it wouldn't fix LE12 and posted the same fix to LE12 subsequently. The latter has been merged and the former might never been merged, depending if a better fix is possible. This however means that the problem is fixed in test builds and most likely in 12.0.2 :crossed_fingers:

XxAcielxX, 2024-09-13:
@NetDwarf

Sorry for late response as I was testing it. Now I can confirm that patch https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/patches/kodi/000-temp-revert-fences.patch fixes my issue. So I'll close this.

closed 2024-09-13 by XxAcielxX