LibreELEC/LibreELEC.tv #9090: [BUG] LE11 Fails to Build: wireguard-tools Checksum Mismatch (sources.libreelec.tv missing package)

xags, 2024-07-13:
### Describe the bug
LE11 fails to build with the error "Incorrect checksum calculated on downloaded" for the wireguard-tools-1.0.20210914.tar.xz package.

### How to reproduce
Steps to reproduce the behavior:
1. git clone https://github.com/LibreELEC/LibreELEC.tv -b libreelec-11.0 
2. cd to source
3. Build the project: PROJECT=Generic THREADCOUNT=35 BUILDER_NAME=xags BUILDER_VERSION=11.06.xags DEBUG=yes VALGRIND=yes ARCH=x86_64 make image
4. See error

### Information
 - LibreELEC Version: LE11, latest
 - Hardware Platform: Generic/Legacy

### Log file
```[282/295] [DONE] install mesa:target
<<< wireguard-tools:target seq 286 <<<
GET      wireguard-tools (archive)
--2024-07-13 10:51:51--  https://git.zx2c4.com/wireguard-tools/snapshot/wireguard-tools-v1.0.20210914.tar.xz
Resolving git.zx2c4.com (git.zx2c4.com)... 86.109.7.149, 2604:1380:45f1:e200::3
Connecting to git.zx2c4.com (git.zx2c4.com)|86.109.7.149|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-xz]
Saving to: '/archive/src/LibreELEC.tv/sources/wireguard-tools/wireguard-tools-1.0.20210914.tar.xz'

          /archive/     [<=>                 ]       0  --.-KB/s               
/archive/src/LibreE     [ <=>                ]  97.45K  --.-KB/s    in 0.06s   

2024-07-13 10:51:52 (1.60 MB/s) - '/archive/src/LibreELEC.tv/sources/wireguard-tools/wireguard-tools-1.0.20210914.tar.xz' saved [99788]

    WARNING      Incorrect checksum calculated on downloaded file: got 04af1ce7b9856b986dad23f7ebd47de7d4e2f48ced32eab058f648d96b801820 wanted 69de713458a887cddb46b2fd1d43bf20548f693136af95792e65ec925538c770
--2024-07-13 10:51:52--  http://sources.libreelec.tv/mirror/wireguard-tools/wireguard-tools-1.0.20210914.tar.xz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 65.109.172.87
Connecting to sources.libreelec.tv (sources.libreelec.tv)|65.109.172.87|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-07-13 10:51:52 ERROR 404: Not Found.

--2024-07-13 10:51:52--  https://git.zx2c4.com/wireguard-tools/snapshot/wireguard-tools-v1.0.20210914.tar.xz
Resolving git.zx2c4.com (git.zx2c4.com)... 86.109.7.149, 2604:1380:45f1:e200::3
Connecting to git.zx2c4.com (git.zx2c4.com)|86.109.7.149|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-xz]
Saving to: '/archive/src/LibreELEC.tv/sources/wireguard-tools/wireguard-tools-1.0.20210914.tar.xz'

          /archive/     [<=>                 ]       0  --.-KB/s               
/archive/src/LibreE     [ <=>                ]  97.45K  --.-KB/s    in 0.06s   

2024-07-13 10:51:52 (1.56 MB/s) - '/archive/src/LibreELEC.tv/sources/wireguard-tools/wireguard-tools-1.0.20210914.tar.xz' saved [99788]

    WARNING      Incorrect checksum calculated on downloaded file: got 04af1ce7b9856b986dad23f7ebd47de7d4e2f48ced32eab058f648d96b801820 wanted 69de713458a887cddb46b2fd1d43bf20548f693136af95792e65ec925538c770
--2024-07-13 10:51:52--  http://sources.libreelec.tv/mirror/wireguard-tools/wireguard-tools-1.0.20210914.tar.xz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 65.109.172.87
Connecting to sources.libreelec.tv (sources.libreelec.tv)|65.109.172.87|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-07-13 10:51:53 ERROR 404: Not Found.

Cannot get wireguard-tools sources : https://git.zx2c4.com/wireguard-tools/snapshot/wireguard-tools-v1.0.20210914.tar.xz
Try later!
*********** FAILED COMMAND ***********
. "${get_handler}"
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/get "${PKG_NAME}"
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/unpack "${PKG_NAME}" "${PARENT_PKG}"
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install wireguard-tools:target has failed!
```

### Context
This built fine a few weeks ago. I'm troubleshooting a nuvoton_cir/kernel issue between LE10 and LE11.
It appears that wireguard-tools attempts to download from sources.libreelec.tv and returns a 404. Browsing the web directory shows it is missing. It then attempts to download from a mirror? (git.zx2c4.com) and receives a checksum error.

heitbaum, 2024-07-13:
sources.libreelec.tv has been updated with the correct SHA256 file

heitbaum, 2024-07-14:
The correct download link looks to be https://git.zx2c4.com/wireguard-tools/snapshot/wireguard-tools-1.0.20210914.tar.xz without the v. it has the same file contents as the v version, but is the one listed with a list file.

closed 2024-07-17 by xags