LibreELEC/LibreELEC.tv #9099: [BUG]

mood-indigo, 2024-07-17:
### Describe the bug
Cannot boot Odroid N2+ device with https://releases.libreelec.tv/LibreELEC-AMLGX.aarch64-12.0.0-odroid-n2.img.gz, but was able to run 
LibreELEC 11.06 with  https://releases.libreelec.tv/LibreELEC-AMLGX.arm-11.0.6-odroid-n2.img.gz, proving that the 12.0 image is corrupted.

### How to reproduce
1. Download https://releases.libreelec.tv/LibreELEC-AMLGX.aarch64-12.0.0-odroid-n2.img.gz
2. Burn image to eMMC module.
3. Insert module in Odroid-N2+ device.
4. Power on device and waited 2 minutes.
Nothing happened.

5. Download https://releases.libreelec.tv/LibreELEC-AMLGX.arm-11.0.6-odroid-n2.img.gz
6. Burn image and power on device to see the LibreELEC setup instructions.

### Information
 - LibreELEC Version: 12.0
 - Hardware Platform: Odroid N2+

### Context
After burning LibreELEC-AMLGX.aarch64-12.0.0-odroid-n2.img.gz to eMMC module and mounting it, I can see the following files:
  - KERNEL
  - KERNEL.md5
  - SYSTEM
  - SYSTEM.md5
  - System Volume Information
    - IndexerVolumeGuid
    - WPSettings.dat
  - amlogic
    - meson-g12a-radxa-zero.dtb
    - meson-g12a-sei510.dtb
    - meson-g12a-u200.dtb
    - ...
    - meson-g12b-odroid-go-ultra.dtb
    - meson-g12b-odroid-n2.dtb
    - meson-g12b-odroid-n2l.dtb
    - ...
    - meson-sm1-x96-air.dtb
  - extlinux
    - extlinux.conf

With the 11.06 image, the `amlogic` directory does not exist and the root directory contains `meson-g12b-odroid-n2-plus.dtb` and `meson-g12b-odroid-n2.dtb`.  This leads me to suspect that the 12.0 image was not created correctly.

chewitt, 2024-07-17:
The image is not "corrupted" but the file path to the dtb files is incorrect. The fix was merged in https://github.com/LibreELEC/LibreELEC.tv/pull/9030 two weeks ago.

Use a current nightly image from https://test.libreelec.tv/12.0/Amlogic/odroid-n2/ .. beware the folder contains old arm images too. You need the latest aarch64 one. Or you can install using LE11 then update to LE12.

closed 2024-07-17 by chewitt

mood-indigo, 2024-07-18:
I tried https://test.libreelec.tv/12.0/Amlogic/odroid-n2/LibreELEC-AMLGX.aarch64-12.0-nightly-20240712-af12d28-odroid-n2.img.gz; but still, no joy.

This image has the same file/directory layout as the 12.0.0 image in the [download page](https://libreelec.tv/downloads/): The various \*.dtb files are still in the `amlogic` directory.

I tried an experiment which wasn't successful.  Since the 11.06 image worked for me, I decided to make the [20240712 image](https://test.libreelec.tv/12.0/Amlogic/odroid-n2/LibreELEC-AMLGX.aarch64-12.0-nightly-20240712-af12d28-odroid-n2.img.gz) have the same layout as the 11.06 image.  I moved the \*odroid\*.dtb to the top directory, removed the `amlogic` directory and edited the `extlinux/extlinux.conf`, changing `FDTDIR` to `/`.

Should I reopen this issue?

mood-indigo, 2024-07-18:
Hi,

I downloaded all the 12.0.0 images on https://libreelec.tv/downloads/amlogic/ to compare them.  They differ only in the `extlinux/extlinux.conf` file; in particularly the difference is only in the value of the `FDT` field (except the Odroid-C4 and Odroid-N2 images do not have this field but the `FDTDIR` field).  For example, the `FDT` value for the NanoPi image is `/amlogic/meson-gxbb-nanopi-k2.dtb` while the value in the WeTek Play2 image is `/amlogic/meson-gxbb-wetek-play2.dtb`.

I am now wondering if the reason why the Odroid N2 image fails to boot is in the build script (that is, the script that builds the images), since the [20240712 image](https://test.libreelec.tv/12.0/Amlogic/odroid-n2/LibreELEC-AMLGX.aarch64-12.0-nightly-20240712-af12d28-odroid-n2.img.gz) has identical content in its `extlinux/extlinux.conf` file.  In other words, I suspect all the nightly-built images for the Odroid N2 will not boot.

mood-indigo, 2024-07-19:
> The image is not "corrupted" but the file path to the dtb files is incorrect. The fix was merged in #9030 two weeks ago.
>
> Use a current nightly image from [...]

@chewitt The nightly images also fail to boot.