LibreELEC/LibreELEC.tv #9116: [RFE] allow use of pip in LE build

heitbaum, 2024-07-22:
### Describe the request

Python packages have started not supporting setuptools. Update build system to allow for PIP.
- #9115

some work was done when we tried to update qemu, but had to be partially rolled back. This is also a blocker to moving to Python 3.12 (or maybe it is better to move to Python12 - which requires refactoring the build system dropping many of the patches that are setuptools based and won’t apply to Python3.12

### Information
 - LibreELEC Version: [e.g. 9.2.1] LE13
 - Hardware Platform: [e.g. RPi3] All

### Log file

### Context

chewitt, 2024-07-22:
@heitbaum see https://github.com/LibreELEC/LibreELEC.tv/pull/4264 for an earlier attempt at pip.

heitbaum, 2024-08-23:
Need pip for Pillow
- https://github.com/python-pillow/Pillow/issues/7734
- https://github.com/LibreELEC/LibreELEC.tv/pull/8536

heitbaum, 2024-08-24:
- Closed resolved by #9226

closed 2024-08-24 by heitbaum