LibreELEC/LibreELEC.tv #9118: [BUG] LE13: runs 2h behind the locale time

eldersixpack13, 2024-07-22:
### Describe the bug
LE13: runs 2h behind the locale time (summertime)

### How to reproduce
Steps to reproduce the behavior:
1. watch a locale clock and the time in the upper right corner of EPG

### Information
 - LibreELEC Version: LE13 (last nightly)
 - Hardware Platform: NUC13

### Log file
none

### Context
- it's afresh/clean installed LE13
- Interface => Regional is proper set (Region default format: "Central Europe")
- time server is set (Network settings)

chewitt, 2024-07-22:
CEST is currently +2 hours on UTC so this "not technicaly a bug" and the expected behaviour. Kodi 22 removed support for setting timezone inside Kodi itself and https://github.com/LibreELEC/service.libreelec.settings/pull/322 hasn't been merged to LE settings yet, so Kodi is correctly showing the (unconfigured/unconfigurable) OS time which is UTC.

antonlacon, 2024-07-23:
You can create a file at `/storage/.cache/timezone` with the following content in the meantime: `TIMEZONE=Europe/Berlin` Look at /usr/share/zoneinfo/Europe if Berlin isn't a nearby city. It probably only needs to restart the tz-data and kodi services, but I'd just reboot to be sure all services take notice.

heitbaum, 2024-08-13:
- Closing as fixed by #9174

closed 2024-08-13 by heitbaum

eldersixpack13, 2024-08-20:
confirming bug is fixed

Thanks