LibreELEC/LibreELEC.tv #9119: [BUG]Build Image not possible with Fedora 40 due wget2

the-dreamer, 2024-07-23:
### Describe the bug
Fedora 40 seems to switch to wget2:

wget --help
GNU Wget2 V2.1.0 - multithreaded metalink/file/website downloader

and the option --passive-ftp has been droped. 

resulting in 

<< wireless-regdb:target seq 5 <<<
GET      wireless-regdb (archive)
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'
Unknown option 'passive-ftp'

Cannot get wireless-regdb sources : https://www.kernel.org/pub/software/network/wireless-regdb/wireless-regdb-2024.07.04.tar.xz
Try later!
*********** FAILED COMMAND ***********

### How to reproduce
switch to wget2 or fedora 40

### Information

### Log file
provided above

### Context
Building image for rpi/x86

heitbaum, 2024-07-23:
https://gitlab.com/gnuwget/wget2/-/issues/661 lot of discussion. Wget2 does not support ftp.

the-dreamer, 2024-07-23:
i see,

I was looking into the LE repo and found only 2 packages which are downloading via ftp. Alsa and libid3tag. 

alsa is also serving source via http, libid3tag via sourcforge. 

is ftp necessary at all?

otherwise i am stuck until ftp is supported in wget2.

heitbaum, 2024-07-23:
There’s some more discussion on fedoras page. https://discussion.fedoraproject.org/t/f40-change-proposal-wget2-as-wget/96422 

I’m assuming that someone has/will fix fedora to return a working wget, and leave the wget2 as-is.

the-dreamer, 2024-07-23:
it is not fixed as i can see. and i see, make libreelec wget2 compatible is not as easy as i thought with avoid ftp downloads. There seems to be more changes. Bad move from fedora team. uff.

henning-schild, 2024-09-05:
Looking at https://gitlab.com/gnuwget/wget2/-/issues/3 it is safe to assume that ftp support in wget2 will not come anytime soon, if at all.
The times of ftp:// are basically over.

We should assume that
 1. wget2 will probably not support ftp anytime soon, if at all
 2. more and more distros will switch from wget to wget2

That brings us to two possible solutions for the future:
 1. switch all remaining ftp:// URIs to https://
 2. switch to `curl` or `lftp` for ftp:// or use `curl` for everything

In the short term affected users can build using a container and https://wiki.libreelec.tv/development/build-docker should be checked/updated to cover `podman` as well as `docker`.

henning-schild, 2024-09-05:
@the-dreamer can you please share the changes you have so far

can you provide some more information on the "addons" which could cause problems. I did never build libreelec, does it pull in additional repos which can contain URIs?

henning-schild, 2024-09-05:
Also note that `busybox wget` on fedora still seems to work. So one could do something like

```
rc=0
wget --passive-ftp ... || rc=$?
# parse error indicating wget2 which lacks ftp support
if [ $rc eq 2 ]; then
  busybox wget --passive-ftp ...
fi
```

the-dreamer, 2024-09-06:
This is the only change i did for fedora 40. that is nothing for the Project as i did it only for me. 

in /script/get_archive
```
diff --git a/scripts/get_archive b/scripts/get_archive
index 626a6f4652..f5a73cb991 100755
--- a/scripts/get_archive
+++ b/scripts/get_archive
@@ -22,7 +22,7 @@ pkg_lock_status "GETPKG" "${PKG_NAME}" "unpack" "downloading package..."
 
 PACKAGE_MIRROR="${DISTRO_MIRROR}/${PKG_NAME}/${PKG_SOURCE_NAME}"
 [ "${VERBOSE}" != "yes" ] && WGET_OPT=-q
-WGET_CMD="wget --output-file=- --timeout=30 --tries=3 --passive-ftp --no-check-certificate -c ${WGET_OPT} --progress=bar:force --show-progress -O ${PACKAGE}"
+WGET_CMD="wget --output-file=- --timeout=30 --tries=3 --no-check-certificate -c ${WGET_OPT} --progress=bar:force --force-progress -O ${PACKAGE}"
 
 # unset LD_LIBRARY_PATH to stop wget from using toolchain/lib and loading libssl.so/libcrypto.so instead of host libraries
 unset LD_LIBRARY_PATH
```

i have not build any extra addon as i am fetching it regulary from libreelec repo.

i have seen, there is only one addon-depend (alsa libid3tag  so far i remember) but it can be changed to http download. In my opinion ftp support can be dropped completly because ftp is practically not used anywhere. 

as of time of my build heimdal needed a patch which was not added from LE13 to LE12

PR #9280 (2024-09-14): scripts and tools: only use --passive-ftp for ftp: urls
All ftp:// urls have been migrated to http/https.

This change maintains the capability to retrieve ftp urls but only adds the —passive-ftp argument to wget if it is a ftp: url.

- fixes #9119
closed 2024-09-14 by CvH