LibreELEC/LibreELEC.tv #9130: [BUG] Some openssl dgst algorithms are enabled but return error

lulol, 2024-07-28:
### Describe the bug
Some openssl dgst algorithms are [enabled](https://github.com/LibreELEC/LibreELEC.tv/blob/3b45f558bc579c505a7461643c984cb56d989f50/packages/security/openssl/package.mk#L16) but return error:

-md4 -mdc2 -whirlpool

### How to reproduce
List enabled digests:
`/usr/bin/openssl dgst -list`

Run test for all enabled digests:
```
for d in $(/usr/bin/openssl dgst -list); do
  test "${d:0:1}" != '-' && continue
   echo "" | openssl dgst $d >/dev/null || echo "*** ERROR $d"
done
```
Results:
```
Error setting digest
40A0EDB47F000000:error:0308010C:digital envelope routines:inner_evp_generic_fetch:unsupported:crypto/evp/evp_fetch.c:355:Global default library context, Algorithm (MD4 : 99), Properties ()
40A0EDB47F000000:error:03000086:digital envelope routines:evp_md_init_internal:initialization error:crypto/evp/digest.c:272:
*** ERROR -md4
Error setting digest
40B0379F7F000000:error:0308010C:digital envelope routines:inner_evp_generic_fetch:unsupported:crypto/evp/evp_fetch.c:355:Global default library context, Algorithm (MDC2 : 98), Properties ()
40B0379F7F000000:error:03000086:digital envelope routines:evp_md_init_internal:initialization error:crypto/evp/digest.c:272:
*** ERROR -mdc2
Error setting digest
40608EA97F000000:error:0308010C:digital envelope routines:inner_evp_generic_fetch:unsupported:crypto/evp/evp_fetch.c:355:Global default library context, Algorithm (whirlpool : 97), Properties ()
40608EA97F000000:error:03000086:digital envelope routines:evp_md_init_internal:initialization error:crypto/evp/digest.c:272:
*** ERROR -whirlpool
```

### Information
`LibreELEC (official): 12.0.0 (RPi4.aarch64)`

### Context
md4 digest was working previously in LE 9.2

antonlacon, 2024-07-28:
Openssl 3 deprecated those algorithms as broken or unused. See https://github.com/openssl/openssl/issues/21247 for example.

They may still be used by adding `-provider legacy` to how they're called, but expect them to be removed in the future.

lulol, 2024-07-28:
Thank you for the explanation and alternative solution.

closed 2024-07-28 by lulol