LibreELEC/LibreELEC.tv #9175: [BUG] LibreELEC 12 (omega) CURL bug/regression breakes WebDAV remote storage

seppiola, 2024-08-12:
### Describe the bug
Curl has been updated in libreelec 12 to version 8.6.0 which introduced [this](https://github.com/curl/curl/issues/13722) regression and as a result the CCurlFile() function called by kodi on library scan and scraping almost always returns an error when called on DAV(s) remote storage like the following:

`CCurlFile::Exists - <davs://server/path/> Failed: Weird server reply(8)`
`Process directory 'davs://server/path/' does not exist - skipping scan.`

making impossible to use webdav as a remote storage on 12 (omega) as reported [here](https://github.com/xbmc/xbmc/issues/25305), [here](https://www.kodinerds.net/thread/78544-nach-update-auf-libreelec-omega-beta-webdav-failed-weird-server-reply-8) and [here](https://forum.libreelec.tv/thread/28684-le-12-0-pi-4b-2gb-unable-to-update-library-compressing-db-0/#post193311).

Curl needs to be bumped at least to version 8.7.1 which should solve the issue.
I confirm from testing different curl versions outside coreelec on my webdav storage that on 8.7.1 libcurl at least doesn’t return any error on GETs, couldn’t try on kodi yet to see the CCurlFile() response.

### How to reproduce
Steps to reproduce the behavior:
1. Add a remote storage using WebDAV
2. Start the library update/scraping
3. Library scan fails with following errors in log

### Information
 - LibreELEC Version: 12.0
 - Hardware Platform: Any, but specifically Odroid N2+

### Log file
```
     info <general>: UpdateLibraries: Starting video library startup scan
     info <general>: VideoInfoScanner: Starting scan ..
    error <general>: CCurlFile::Exists - <davs://webdav.server.adress:443/remote/path/1/> Failed: Weird server reply(8)
  warning <general>: Process directory 'davs://webdav.server.adress:443/remote/path/1/' does not exist - skipping scan.
    error <general>: CCurlFile::Exists - <davs://webdav.server.adress:443/remote/path/2/> Failed: Weird server reply(8)
  warning <general>: Process directory 'davs://webdav.server.adress:443/remote/path/2/' does not exist - skipping scan.
    error <general>: CCurlFile::Exists - <davs://webdav.server.adress:443/remote/path/3/> Failed: Weird server reply(8)
  warning <general>: Process directory 'davs://webdav.server.adress:443/remote/path/3/' does not exist - skipping scan.
    error <general>: CCurlFile::Exists - <davs://webdav.server.adress:443/remote/path/4/> Failed: Weird server reply(8)
  warning <general>: Process directory 'davs://webdav.server.adress:443/remote/path/4/' does not exist - skipping scan.
    error <general>: CCurlFile::Exists - <davs://webdav.server.adress:443/remote/path/5/> Failed: Weird server reply(8)
  warning <general>: Process directory 'davs://webdav.server.adress:443/remote/path/5/' does not exist - skipping scan.
     info <general>: VideoInfoScanner: Finished scan. Scanning for video info took 857 ms
```

### Context
Nothing particular, a simple fresh install presents the issue.

seppiola, 2024-08-14:
@heitbaum thanks, I built an image myself and will test it in the coming days.

This probably isn't the place to ask about this... but how do you deal with binary packages tarballs being removed from upstream mirrors? Store them along the toolchain? I had to bump a couple of packages versions to have libreelec built with the patch from #9178, hope this won't break things.

In the meantime I'had already built and tested a coreelec 21 (omega) image, since I had that toolchain ready, with updated curl and can confirm the issue is gone, at least as far as I can tell.

Will let you know as soon as I can if this patch also fixes the issue.

heitbaum, 2024-08-14:
> @heitbaum thanks, I built an image myself and will test it in the coming days.
>
>
>
> This probably isn't the place to ask about this... but how do you deal with binary packages tarballs being removed [...]

If you see some missing upstream package (Omega/Piers) - please reach out. And we will add them to our sources server so they will get picked up (or fix the PKG_URL.)

seppiola, 2024-08-15:
> @seppiola can you please confirm that this fixes your reported issue? If you need an image to test with, please let us know what target you need.

I confirm curl 8.6.0 works as intended with this patch and library scan now always works on my setup with webdav remote storage.
I can't seem to find any error in `CCurlFile()` responses in logs.

> If you see some missing upstream package (Omega/Piers) - please reach out. And we will add them to our sources server so they will get picked up (or fix the PKG_URL.)

I noticed I built from a git release tag for 12.0 instead of the actual 12 branch, so most of them had already been fixed and the last one was just updated a couple of days ago, so sorry for the unneeded warning.

closed 2024-08-16 by heitbaum