LibreELEC/LibreELEC.tv #9179: [BUG] Regression in docker building libreelec-12.0

ned14, 2024-08-13:
### Describe the bug
This works with branch `libreelec-10.0` and `libreelec-11.0` but fails in branch `libreelec-12.0`

### How to reproduce

```
git clone https://github.com/LibreELEC/LibreELEC.tv.git
cd LibreELEC.tv
git checkout libreelec-12.0
docker build --pull -t libreelec tools/docker/noble
docker run --rm --log-driver none -v `pwd`:/build -w /build -it libreelec make image
./scripts/image mkimage
mkdir: cannot create directory '/build/build.LibreELEC-Generic.x86_64-12.0-devel': Permission denied
*********** FAILED COMMAND ***********
. config/options ""
**************************************
make: *** [Makefile:10: image] Error 1
```
Previously it created `build.LibreELEC-Generic.x86_64-11.0-devel` just fine.

Note my docker is configured to run from a normal user, not just root.

I note that under `libreelec-11.0` the files created in the build directory are my non-root user, whereas if you manually create `build.LibreELEC-Generic.x86_64-12.0-devel` it seems to assign uid 1001 before git blows up about user ids smelling fishy.

vpeter4, 2024-08-13:
Try adding `-u $(id -u)`.

ned14, 2024-08-13:
Yes that appears to be working. The documentation needs to be updated. Thanks for the tip.

closed 2024-08-13 by ned14