LibreELEC/LibreELEC.tv #9180: [BUG] Building aom package fails on master

vpeter4, 2024-08-13:
### Describe the bug

Building aom package fails on master and because of it also ffmpegx and tvheadend.

```
[306/309] Building C object CMakeFiles/aom_av1_common_sve2_intrinsics.dir/av1/common/arm/highbd_compound_convolve_sve2.c.o
FAILED: CMakeFiles/aom_av1_common_sve2_intrinsics.dir/av1/common/arm/highbd_compound_convolve_sve2.c.o 
/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/toolchain/bin/ccache /data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/toolchain/bin/aarch64-libreelec-linux-gnu-gcc  -I/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1 -I/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1/.aarch64-libreelec-linux-gnu -I/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1/apps -I/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1/common -I/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1/examples -I/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1/stats -I/data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1/third_party/libyuv/include -march=armv8.2-a+crc+crypto -mabi=lp64 -Wno-psabi -mtune=cortex-a76 -mno-outline-atomics -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG -fPIC -DPIC -DDUMMYOPT -march=armv9-a+sve2 -MD -MT CMakeFiles/aom_av1_common_sve2_intrinsics.dir/av1/common/arm/highbd_compound_convolve_sve2.c.o -MF CMakeFiles/aom_av1_common_sve2_intrinsics.dir/av1/common/arm/highbd_compound_convolve_sve2.c.o.d -o CMakeFiles/aom_av1_common_sve2_intrinsics.dir/av1/common/arm/highbd_compound_convolve_sve2.c.o -c /data/LibreELEC/build.LibreELEC-RPi5.aarch64-13.0-devel/build/aom-3.9.1/av1/common/arm/highbd_compound_convolve_sve2.c
during RTL pass: expand
In file included from ../av1/common/arm/highbd_compound_convolve_sve2.c:19:
In function 'aom_tbl_s16',
    inlined from 'convolve4_4_x' at ../av1/common/arm/highbd_compound_convolve_sve2.c:106:33,
    inlined from 'highbd_dist_wtd_convolve_x_4tap_sve2.isra' at ../av1/common/arm/highbd_compound_convolve_sve2.c:162:23:
../aom_dsp/arm/aom_neon_sve_bridge.h:52:10: internal compiler error: Segmentation fault
   52 |   return svget_neonq_s16(svtbl_s16(svset_neonq_s16(svundef_s16(), s),
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   53 |                                    svset_neonq_u16(svundef_u16(), tbl)));
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
0x7f704578708f ???
	/build/glibc-LcI20x/glibc-2.31/signal/../sysdeps/unix/sysv/linux/x86_64/sigaction.c:0
0x7f7045768082 __libc_start_main
	../csu/libc-start.c:308
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
```

### How to reproduce
Steps to reproduce the behavior:
`PROJECT=RPi DEVICE=RPi5 ARCH=aarch64 ./scripts/build aom`

### Information
 - LibreELEC Version: 13.0
 - Hardware Platform: RPi5

### Log file

### Context

This gcc commit fixes the issue:
https://github.com/gcc-mirror/gcc/commit/0970ff46ba6330fc80e8736fc05b2eaeeae0b6a0

More info about it here:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=115464

heitbaum, 2024-08-17:
@vpeter4 I have seen it too. Just haven’t been able to put my finger on it yet. My build tree backlog is simplifying now I have got a few things out of the way. So I would like to tests a full clean aom build on master with gcc-14.1/14.2.

I have done some builds with 14.1 and that does seem to work for aarch64.
With 14.2 and aarch64 - it does fail.

vpeter4, 2024-08-17:
Actually it works right now. Looks like gcc-14.2.0 fix it.

PR #9200 (2024-08-18): gcc: 14.2 upstream fix for building libaom on arm64
- Fixes #9180 

ref:
- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=115464
- https://gcc.gnu.org/git/gitweb.cgi?p=gcc.git;h=32b21292adb6ad6b5e1d60d923a773e4d0daca7b
- https://gcc.gnu.org/git/gitweb.cgi?p=gcc.git;h=86dacfb06b90371458d58872f461d358a0834305
- https://gcc.gnu.org/git/?p=gcc.git;a=shortlog;h=refs/heads/releases/gcc-14
closed 2024-08-18 by jernejsk