LibreELEC/LibreELEC.tv #9181: [BUG] No sound out of the 3.5mm jack on Pi 3

JonnyGators, 2024-08-13:
### Describe the bug

When playing back music, there is no sound output the 3.5 mm jack on a Raspberry Pi 3

### How to reproduce
Steps to reproduce the behavior:
Setup on Pi 3
Try to play music
Hear the silence

### Information
 - LibreELEC Version: 12.0.0
 - Hardware Platform: Raspberry Pi 3

### Log file

### Context

chewitt, 2024-08-14:
The support forum is here: https://forum.libreelec.tv

closed 2024-08-14 by chewitt

JonnyGators, 2024-08-14:
The complete lack of audio output working for an app that is designed to output audio is a glaring bug, and puzzling choice.  Even more puzzling is closing out such a glaring bug with no explanation.

I've yet to encounter a more stupid scenario than an audio player that is designed to not output audio.

chewitt, 2024-08-14:
It's a config issue requiring a one-line change not a bug.