LibreELEC/LibreELEC.tv #9244: [BUG] LibreELEC 12.0.1 unbootable on Rock64 (RK3328)

NetDwarf, 2024-08-27:
### Describe the bug
A fresh install of LibreELEC 12.0.1 is stuck on boot. LibreELEC logo and version shows, but it remains on that screen forever.

### How to reproduce
Steps to reproduce the behavior:
1. Flash LibreELEC 12.0.1 (stable) to SD card
2. Insert SD
3. Start device

### Information
 - LibreELEC Version: 12.0.1
 - Hardware Platform: Rock64 (RK3328)

### Log file
Unfortunately there is no Kodi.log or anything useful I could find, so you might need to point me in the right direction. 

I can say that the installation process seemingly went fine as the partitions are there and have the right size as far as I can tell.

### Context
This is also true for newer nightly releases (after 20240803 at least)

heitbaum, 2024-08-27:
- Please check if this is the same as #8446

NetDwarf, 2024-08-28:
> * Please check if this is the same as [Since #7837 kernel 6.4.y Rock64 does not display on a 4K screen #8446](https://github.com/LibreELEC/LibreELEC.tv/issues/8446)

It's not. It's on a 1080p display, kernel 6.6, 12.0.1 stable.

heitbaum, 2024-08-29:
I had a look - you are right - only the kodi and kernel really different between these 2 

https://github.com/LibreELEC/LibreELEC.tv/compare/cbf0545...12.0.1

Did you get anything on the screen during the 12.0.1 boot?

I'm going to suggest 
- boot with a clean https://test.libreelec.tv/12.0/Rockchip/rock64/LibreELEC-RK3328.aarch64-12.0-nightly-20240802-cbf0545-rock64.img.gz image. 
- validate
- put the 12.0.1 image into the .update folder

NetDwarf, 2024-08-29:
> I had a look - you are right - only the kodi and kernel really different between these 2
> 
> [cbf0545...12.0.1](https://github.com/LibreELEC/LibreELEC.tv/compare/cbf0545...12.0.1)
> 
> Did you get anything on the screen during the 12.0.1 boot?
> 
> I'm going to suggest
> 
>     * boot with a clean https://test.libreelec.tv/12.0/Rockchip/rock64/LibreELEC-RK3328.aarch64-12.0-nightly-20240802-cbf0545-rock64.img.gz image.
> 
>     * validate
> 
>     * put the 12.0.1 image into the .update folder

* 12.0-nightly-20240802 boots and then updating to 12.0.1 -> broken
* 12.0.0 (stable) boots and then updating to 12.0.1 -> broken

I think it is safe to say, that 12.0.1 is broken on this specific setup at least (Rock64 RK3328).

Is there actually a way to debug this stuck boot somehow?

heitbaum, 2024-08-30:
We are checking - it might be fixed with an upstream patch - https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/pinctrl/pinctrl-rockchip.c?id=128f71fe014fc91efa1407ce549f94a9a9f1072c - will advise once tested.

diederikdehaas, 2024-08-31:
FWIW: That is a known fix for a known bug

(https://lore.kernel.org/all/2241778.ZfL8zNpBrT@diego/ was send after I 'pushed' on the issue)

heitbaum, 2024-08-31:
@NetDwarf please test https://heitbaum.libreelec.tv/LibreELEC-RK3328.aarch64-12.0-devel-20240831103135-ee40b52-rock64.img.gz and let us know, then we can include the patch in the nightly releases if it does indeed fix your issue.

diederikdehaas, 2024-08-31:
@heitbaum How can I get serial output? I downloaded the above image and the 12.0.1 release and have a serial console connected to my Rock64, but when I started up, I didn't get anything on serial.  
How to change that?

diederikdehaas, 2024-08-31:
Surprisingly (?) the serial output was the same, apart from timestamps/guids and the release identifier.

I'll now try the visual approach, but that's without serial. Could it be that this issue doesn't happen always? Or only with certain HW revisions? I have a Rock64 v2

NetDwarf, 2024-09-03:
> @NetDwarf please test https://heitbaum.libreelec.tv/LibreELEC-RK3328.aarch64-12.0-devel-20240831103135-ee40b52-rock64.img.gz and let us know, then we can include the patch in the nightly releases if it does indeed fix your issue.

Updating from 12.0.1 (offline) worked, so I'd assume it just works. 

PS.: Sorry for the late feedback, but I haven't had access to the device in the last few days.

closed 2024-09-07 by NetDwarf