LibreELEC/LibreELEC.tv #9276: [BUG] Libespot 20.0.130 zoomed in artwork display

bdgra, 2024-09-08:
Hi,

I'm still using Librespot 20.0.130 on a Kodi system. Reason is that from version 20.5 and later 21.1.1. Discovery mode isn't working anymore and it stops after the album has been finished. With 20.0.130 this all works fine. However the artwork of an album has been zoomed in and isn't clearly visible. Is there a way to solve this? I only have this with this addon. And already tried to change the system resolution. But unfortunately without any luck. 

Hope someone can help with this matter.

heitbaum, 2024-09-08:
> Hi,
>
> I'm still using Librespot 20.0.130 on a Kodi system. Reason is that from version 20.5 and later 21.1.1. Discovery mode isn't working anymore and it stops after the album has been finished. With [...]

I’d suggest reaching out on the https://github.com/librespot-org/librespot git for the problems you are experiencing, as the build of the addon is just a straight compile of librespot. If you can identify a fix … update that fixes the issue, we can update the addon based on that release of librespot. @henri-funck made some updates last year to the version.

bdgra, 2024-09-08:
@henri-funck do you have a solution for this version? Hope you can help

heitbaum, 2024-10-29:
@bdgra did you have success with the librespot team? Can you add the xref ticket? Or can we close this?

heitbaum, 2024-11-02:
If still an issue. Please raise on the forums. Can be reopened if there is anything else we can do on the LE side to fix.

closed 2024-11-02 by heitbaum