LibreELEC/LibreELEC.tv #9305: [BUG] net-snmp spam journald

prahal, 2024-09-16:
### Describe the bug

Since 6.7 unpatched snmp outputs repeatedely:
`systemstats_linux: unexpected header length in /proc/net/snmp. 237 != 224`

Please include  [Add Linux 6.7 compatibility parsing /proc/net/snmp](https://github.com/net-snmp/net-snmp/pull/785)  to https://github.com/LibreELEC/LibreELEC.tv/blob/HEAD/packages/addons/service/net-snmp/package.mk#L4
Though it should go to the addin upstream too, thus I will open a bug report there too asap.

### How to reproduce
Steps to reproduce the behavior:
1. Instal net-snmp kodi plugin
2. PCHeck journalctl output
3. See error

### Information
 - LibreELEC Version: nightly-20240914-7b7b3db
 - Hardware Platform: Odroid-C2

### Log file

https://paste.libreelec.tv/crack-mantis.log
```
Sep 16 04:30:26 tv snmpd[849]: systemstats_linux: unexpected header length in /proc/net/snmp. 237 != 224
```
### Context

closed 2024-09-17 by CvH

prahal, 2024-09-17:
Thanks a lot!
Could it be backported to the LE 12 branch?

heitbaum, 2024-09-17:
> Thanks a lot! Could it be backported to the LE 12 branch?

LE12 uses Kernel 6.6 - so the patch is not required.

prahal, 2024-09-17:
I mean for LE12 nightly
```
uname -a
Linux tv 6.8.0 #1 SMP PREEMPT Sat Sep 14 09:31:32 UTC 2024 aarch64 GNU/Linux
```

```
cat /etc/os-release 
NAME="LibreELEC"
VERSION="nightly-20240914-7b7b3db"
ID="libreelec"
VERSION_ID="12.0"
PRETTY_NAME="LibreELEC (community): nightly-20240914-7b7b3db"
HOME_URL="https://libreelec.tv"
BUG_REPORT_URL="https://github.com/LibreELEC/LibreELEC.tv"
BUILD_ID="7b7b3dbe828d08994a551ba33c7886861a44630e"
LIBREELEC_ARCH="AMLGX.aarch64"
LIBREELEC_BUILD="community"
LIBREELEC_PROJECT="Amlogic"
LIBREELEC_DEVICE="AMLGX"
```

PR #9306 (2024-09-17): net-snmp: fix journald warnings and update addon (1)
- fixes #9305