LibreELEC/LibreELEC.tv #9319: [BUG] Method not implemented error when using jsonrpc

devhl-labs, 2024-09-21:
### Describe the bug
I just migrated from an old version of libreelec to this version https://releases.libreelec.tv/LibreELEC-RPi4.aarch64-12.0.0.img.gz
Now when I make a jsonprc request, I get a 501 error. Am I doing something wrong, or is this a bug?

```javascript
const showNotification = {
    "jsonrpc": "2.0",
    "method": "GUI.ShowNotification",
    "params": {
        "title": "AutoPi",
        "message": `Playing: ${request.play}`,
        "displaytime": 15000
    },
    "id": 1
};

const url = `http://${ip}:${port}/jsonrpc`;

fetch(url, {
    method: 'POST',
    body: JSON.stringify(showNotification),
    headers: {
        'content-type': 'application/json'
    }
})
    .then(function (response) {
        console.log(reponse);                // 501
        return response.json();
    })
}
```

DaVukovic, 2024-10-20:
I've just tested against a freshly compiled Kodi on Fedora with: 

```
curl -X POST -H "content-type:application/json" http://127.0.0.1:8080/jsonrpc -u kodi:xxxx -d '{"jsonrpc":"2.0","id":1,"method":"GUI.ShowNotification","params":{"title":"AutoPi","message":"Playing","dosplaytime":15000}}'
```

And that worked fine

So, I'm a bit confused about the 501 message which, AFAICT, means an unsupported request. And that, obviously, is not the case.

devhl-labs, 2024-10-20:
Thank you so much for trying. I fixed it by enabling the web server and switching ports. Sorry i forgot to close this.

closed 2024-10-20 by devhl-labs