LibreELEC/LibreELEC.tv #9324: [BUG] sound from streamed sources stutters after a minute

mateuszviste, 2024-09-24:
### Describe the bug
I have installed the radio plugin from https://github.com/fivebanger/kodi_addons/tree/master/plugin.audio.radiode on latest version of librelec. The plugin works, is hardly usable since when I start a radio stream it works fine for about half a minute, and then starts to stutter. Meaning: about every 2s there is a very short (like 0.1s) silence gap.

I've looked how the plugin works and I see that it does no sound-related stuff at all, just calls some xbmc functions to play an audio stream, so I suppose the problem is not with the plugin itself but more general.

This runs on an RPi 4 which plays local MP3 files fine, as well as movies from a local NFS drive, so it does not seem to be hardware-related.

My internet connection is 200Mbps+ with a low latency. Playing a radio stream from my laptop via the radio.de website also works fine, so it does not seem to be network-related.

The RPi4 issue sounds like a buffering problem - like maybe the application buffers 2s of sound and then needs a short moment to fetch the next 2s. That's just a subjective guess, of course. Or maybe it plays the streamed sound slightly too fast, hence it runs out of data to be played and must way for the stream station to produce some new samples...

I have tried to fiddle with the UI configuration: disabled/enabled caching and changing the expert sound settings (optimized / best match / fixed), but nothing helped so far.

Is there anything else I could try to improve the sound quality?

### Information
 - LibreELEC Version: LibreELEC (official): 12.0.1 (RPi4.aarch64)
 - Hardware Platform: RPi4

chewitt, 2024-09-24:
Please report this to the addon author via the Kodi forums. LE staff have no knowledge of the addon code, and if as you say it's just playing an audio stream, it's unlikely to be something LE specific.

closed 2024-09-24 by chewitt

mateuszviste, 2024-09-24:
Thank you for your suggestion. In the meantime, after many different tests, I finally located the problem. As I was saying, I am using an RPi4. What I did not say (because I did not think it was important) is that I am using its analog output for sound.
When I was setting up this RPi I noticed that sound does not work, and after reading CONFIG.TXT I found that for some reasons it is on purpose and that I had to uncomment the following lines:
````
# uncomment to enable analog audio output
#dtparam=audio=on
#audio_pwm_mode=1
````

I did so and - indeed - sound worked then. Why it is disabled by default is a mystery to me, but I did not think much of it. At that time I was not using streaming audio services, and everything else worked well, so I thought it is all good.

Now back to the problem of stuttering audio when listening to online radios: I found out today that if I comment out the line `audio_pwm_mode=1` then the online radios suddenly work. All other things I use (movies & mp3 files from NFS storage) also still work fine. I have no idea why this setting impacts specifically online audio streaming and not local playback, but for some reason it just does. I have also no clue why it is (implicitly) recommended to uncomment it when enabling analog audio. Surely there are good reasons for all this, but for my case it was not a good configuration.

So, to sum it up shortly: in my situation (LE12 + RPi4 + analog audio output) it is necessary to uncomment `dtparam=audio=on` in CONFIG.TXT but leave `audio_pwm_mode=1` commented out. And now it all works perfectly.