LibreELEC/LibreELEC.tv #9337: [BUG] Orange Pi PC won't boot.  Has been broken for a while

NickelNuts, 2024-09-30:
This has been discussed in the forum here:  https://forum.libreelec.tv/thread/28364-compiling-from-master-doesn-t-generate-bootable-orange-pi-pc-images-any-more/?postID=190326&highlight=orange%2Bpi%2Bpc#post190326

I created this post here because it looks like the post from August 14, 2024, describes a simple fix to the u-boot patches.

Sorry, my technical skills are not great, but I do have a Orange PI PC I could test an image on

jernejsk, 2024-10-01:
Can you try fix in #9343?

NickelNuts, 2024-10-02:
> Can you try fix in #9343?

Does this get reflected in the nightly builds, or should I set up a cross-compiler, and build?

jernejsk, 2024-10-02:
You have to build it yourself using command `PROJECT=Allwinner ARCH=arm DEVICE=H3 UBOOT_SYSTEM=orangepi-pc make image`. See https://wiki.libreelec.tv/development/build-basics for how to set up environment.

NickelNuts, 2024-10-02:
OK thanks.  I have  the compiler running.  I manually deleted 0011-DO-NOT-MERGE-sunxi-Enable-remoteproc-on-some-H3-boar.patch and added 0011-DO-NOT-MERGE-sunxi-Enable-remoteproc-on-some-H3.patch

NickelNuts, 2024-10-02:
Now I see I should have built for version 13 instead of 12,  I can redo it if it makes a difference

closed 2024-10-03 by jernejsk