LibreELEC/LibreELEC.tv #9354: [BUG]

OriginaldCrypt, 2024-10-04:
### Describe the bug
Ethernet network adapter problem after power-suspend-resume cycle in NUC 10i3FNH in several versions of LE.

### How to reproduce
Steps to reproduce the behavior:
1. Just powerup the NUC and boot into LE
2. Configure suspend after xx seconds
3. Wait for suspend
4. Wake up the NUC
5. Check ethernet network connection or try to access a network resource

### Information
 - LibreELEC Version: Stable 12.0.1 fails
 - All LibreELEC nightly 12.0 Versions published so far (latest as of today is LibreELEC-Generic.x86_64-12.0-nightly-20240916-05c2a6a.img.gz): fail
 - All LibreELEC v13.0 nightlies up to f79f76a (LibreELEC-Generic.x86_64-13.0-nightly-20240919-f79f76a.img.gz): fail
 - All LibreELEC v13.0 nightlies from 8544bcd (LibreELEC-Generic.x86_64-13.0-nightly-20240920-8544bcd.img.gz): work
 - Hardware Platform: NUC 10i3FNH
 - LE flavor: Generic

### Log file
LibreELEC-Generic.x86_64-13.0-nightly-20240920-8544bcd.img.gz log file (network works after power-suspend-resume cycle):
LibreELEC:~ # lsb_release
LibreELEC (community): nightly-20240920-8544bcd

https://paste.libreelec.tv/witty-hamster.log

### Log file
LibreELEC-Generic.x86_64-13.0-nightly-20240919-f79f76a.img.gz log file (network does not work after power-suspend-resume cycle):
LibreELEC:~ # lsb_release
LibreELEC (community): nightly-20240919-f79f76a

https://paste.libreelec.tv/talented-piranha.log

### Log file
LLibreELEC-Generic.x86_64-12.0.1.img.gz file (network doesn't work after power-suspend-resume cycle):
LibreELEC:~ # lsb_release
LibreELEC (official): 12.0.1

https://paste.libreelec.tv/clever-bull.log

### Context
N/A

heitbaum, 2024-10-04:
Thanks for testing and advising that it is now fixed in LE13/master. 

The LE13 changes (platform/kernel/python and many others) won’t be backported to LE12.

I would suggest tracking the LE13 developments as the draft PR #9348 will probably be the kernel LTS release that LE13 uses when it is released - and testing your issue.

mglae, 2024-10-14:
Connman was updated for LE12 please retest a recent LE12 nightly

OriginaldCrypt, 2024-10-16:
> Connman was updated for LE12 please retest a recent LE12 nightly

It worked!

heitbaum, 2024-10-16:
Closing as fixed.

closed 2024-10-16 by heitbaum