LibreELEC/LibreELEC.tv #9375: [BUG] High load on Samba server in ver 12 and 13

plvlnisse, 2024-10-08:
### Raspberry Pi4 high SMB load on LibreElec V12 and v13???

I have run KibreElec for several years with the media files on an external SMBserver which also is the MySQL server for the databases.

This has worked fine, but recently I have noticed a high load on the SMBserver. I did not notice it when on ver 11.8 and I am not sure if it appeared in ver 12.0 or 12.0.1. i have tried the verc13 nightly buildscandvthey are showing the same high losd

I can see the ethernet light on the RPi flickering all the time, which it didn't do earlier and the smb server thread on then NAS is running 100% CPU (on one of the cores).

No negative effect on either scanning the library or serving streams to the RPi that I can detect.

I have tried enabling debug information in LibreElec, but having problem with that. I get no SMB entries in the log file even when I try to enable both debug info and SMB details in the user GUI.

* Is there any resource describing how I can get info from tte system to try to find out what is going on?

* Anyone else that have seen changed behavior with files hosted on an external SMB host?

heitbaum, 2024-10-29:
@plvlnisse best to raise this on the forum to get support.

closed 2024-10-29 by heitbaum

plvlnisse, 2024-10-29:
Thanks, vill do so.