LibreELEC/LibreELEC.tv #9379: [BUG]Hardware accelleration decoding AV1 on AMD 780M

KarstenL680, 2024-10-09:
### Describe the bug
AV1 Videos decode in Software with high CPU Usage, this is a known issue with MESA Versions older than 24.1.2, so please update MESA to 24.1.2 or newer on Libreelec 12. This issue is also documented at Fedora Linux.

### How to reproduce
Steps to reproduce the behavior:
Just play AV1 encoded Video on AMD 780M based hardware

### Information
 - LibreELEC Version: 12.01
 - Hardware Platform: AMD 7840HS based MiniITX Mainboard with AMD 780M Graphic

### Log file

### Context

heitbaum, 2024-10-09:
Hi @KarstenL680 - LE12 was released with Mesa 24.0 and there are no plans to update Mesa in this stable release. Our master branch which will become LE13 is current with Mesa 24.2.

closed 2024-10-09 by heitbaum

KarstenL680, 2024-10-09:
OK, any information how i can update it on my self?

heitbaum, 2024-10-09:
Its not a clean or simple "update this"; as the dependencies for 24.1 changed the build.

But if you were trying to backport to your self build 12.x image then these 2 PRs are the main ones to incorporate into your image.
- #8926
- #9103

The full list of mesa changes are: https://github.com/LibreELEC/LibreELEC.tv/pulls?q=is%3Apr+is%3Aclosed+mesa. There could be other updates / merge issues as LLVM and others were updated seperately.