LibreELEC/LibreELEC.tv #9414: [BUG]  sundtek-mediatv: wget missing

heitbaum, 2024-10-19:
```
<<< sundtek-mediatv:target seq 63 <<<
BUILD      sundtek-mediatv (target)
    TOOLCHAIN      manual
/build/packages/addons/driver/sundtek-mediatv/package.mk: line 38: wget: command not found
FAILURE: scripts/build sundtek-mediatv:target during make_target (package.mk)
*********** FAILED COMMAND ***********
wget -O installer.tar.gz ${INSTALLER_URL}
**************************************
FAILURE: scripts/build sundtek-mediatv:target has failed!

The following log for this failure is available:
  /build/build.LibreELEC-ARMv7.arm-13.0-devel/.threads/logs/63.log

>>> sundtek-mediatv:target seq 63 >>>
[059/652] [FAIL] build   sundtek-mediatv:target

The following log for this failure is available:
  /build/build.LibreELEC-ARMv7.arm-13.0-devel/.threads/logs/63.log
```

vpeter4, 2024-10-19:
Is it too hard to open this file and change wget to curl? And issue is really need to be open?

CvH, 2024-10-19:
To open an issue not forget about a bug is likely not the worst idea.

vpeter4, 2024-10-19:
> To open an issue not forget about a bug is likely not the worst idea.

That's true. But fix in this case is trivial which means instead PR should be made.

PR #9418 (2024-10-19): sundtek-mediatv: change wget to curl
Closes #9414
closed 2024-10-19 by heitbaum