LibreELEC/LibreELEC.tv #9465: [BUG] Playback of SVTPlay fails with 403 on LE12/RPi5 (but works fine on ubuntu/amd64)

gandalth, 2024-11-04:
### Describe the bug
The plugin SVTPlay allows for watching content from Swedish broadcaster SVT on kodi.
Using Libreelec on RPi5, many shows fails playback and kodi log shows that a 403 http error code has been received.
Surprisingly, playback using the same kodi version on Ubuntu (amd64) works fine. 

### How to reproduce
Steps to reproduce the behavior:
1. Install plugin SVTPlay. Note: much of the content is geoblocked to Sweden.
2. Play any episode of the show Bluey (or Rubble and Crew)
3. See error in kodi.log.

### Information
 - LibreELEC Version: 12.1
 - SVTPlay 5.1.21-matrix1 (installed through addon-browser)
 - Hardware Platform: RPi5

### Log file
2024-10-12 15:13:37.649 T:844 info : VideoPlayer::OpenFile: plugin://plugin.video.svtplay/?mode=video&id=%2Fvideo%2FKrQdGzq%2Fbluey%2Fhinderbana
2024-10-12 15:13:38.211 T:977 error : CCurlFile::Stat - <https://svt-vod-secure-1a.akamaized.net/d0/secure/20240930/a38f8a8d-687b-433c-b6f2-310bd9749fc9/hls-ts-
avc.m3u8> Failed: HTTP response code said error(22)
2024-10-12 15:13:38.211 T:977 info : Creating InputStream
2024-10-12 15:13:38.524 T:977 error : CCurlFile::CReadState::FillBuffer - (0x7ffeec0011d0) Failed: HTTP returned code 403
2024-10-12 15:13:38.524 T:977 error : CCurlFile::Open - <https://svt-vod-secure-1a.akamaized.net/d0/secure/20240930/a38f8a8d-687b-433c-b6f2-310bd9749fc9/hls-ts-
avc.m3u8> Failed with code 403:

### Context
I have discussed this issue with the author of SVTPlay.
https://github.com/kodi-svtplay/xbmc-svtplay/issues/294

gandalth, 2024-11-23:
Any idea on this issue? 
As the svtplay plugin is the same for LE12 and ubuntu, it must be an underlying library which behaves differently.

Any other tests that I should run?

chewitt, 2024-11-23:
The support forum is over here: https://forum.libreelec.tv .. create a post and provide a proper debug log-file, else we have zero information to work with (snippets are rarely/never informative).

closed 2024-11-23 by chewitt