LibreELEC/LibreELEC.tv #9529: [BUG] WeTek Play 2: Black screen

sideeffect42, 2024-11-24:
### Describe the bug
All nightly builds of LibreELEC 13.0 for WeTek Play 2, starting with nightly-20241116-a2cf194, produce only a black screen after Kodi starts.
The boot splash is displayed correctly.

nightly-20241115-7be68a2 is the last working version.

May be related to #9526.

### How to reproduce
Steps to reproduce the behavior:
1. Upgrade to a nightly build of LibreELEC 13.0 from 2024-11-16 or later.
2. Boot the device.
3. See black screen.

### Information
 - LibreELEC Version: 13.0 (nightly builds 2024-11-16 and later)
 - Hardware Platform: wetek-play2

chewitt, 2024-11-24:
Will be fixed by #9526 or a Kodi bump once https://github.com/xbmc/xbmc/pull/25981 has been merged. You can update to this until either of those happen: https://chewitt.libreelec.tv/testing/LibreELEC-AMLGX.aarch64-12.80.0.tar

closed 2024-11-26 by CvH

sideeffect42, 2024-12-07:
The bug is now resolved in the latest nightly builds. Thank you.